skiffos/SkiffOS #219: Mounting an XFS drive on a SkiffOS system

peterwilli, 2022-07-21:
Hey there,

Sorry for making another issue. I tried to mount a XFS HDD on my SkiffOS system, but I get the following error:
```
root@TheBestGhoster:~# mount /dev/sda1 mnt
mount: /root/mnt: unknown filesystem type 'xfs'.
       dmesg(1) may have more information after failed mount system call.
```

Now it's obvious that the module simply isn't there, so I tried to add `CONFIG_XFS_FS=y` to `overrides/buildroot/fs` which was to my best knowledge what I should have done.

This sadly didn't work, and it also seems to have killed other drivers like my wireless driver....

I couldn't find anything online about adding XFS modules, tried looking for both buildroot and SkiffOS, while buildroot has a xfsprogs package, this is just the utilities, not the module to mount these systems...

paralin, 2022-07-21:
> Sorry for making another issue

Make as many as you need!

I enabled the XFS kernel module: 0e83fcfbb66be9f95135d0ba9d883b556d663d9a

Be sure to fully rebuild the kernel if you believe your kernel config changes might need to rebuild some code files (like changing from built-in to modules)

```
git pull
rm -rf ./workspaces/default/build/linux-custom
make configure compile
```

Please remove your `overrides` file for this, since building as a module is less likely to cause problems than `=y` (built in)

Does this solve your issue?

peterwilli, 2022-07-21:
You reply so fast omg! Thanks, I'll try it. Can I ask why it has to be included in configs-base? Wouldn't that mean everyone would get the module, even those that don't need it?

paralin, 2022-07-21:
Because it's a module, it will only be loaded on-demand when needed. So adding more modules like filesystems which could be useful to some users is not a problem.

peterwilli, 2022-07-21:
> Because it's a module, it will only be loaded on-demand when needed. So adding more modules like filesystems which could be useful to some users is not a problem.

Thanks, and just for my idea of what you can and can't put as override, would this work as override as well? Like if I were to place `CONFIG_XFS_FS=m` in my `fs` override would it have the same effect?

paralin, 2022-07-21:
@peterwilli Yes, because `overrides` is just another configuration package which is always at the end of the `SKIFF_CONFIG` list automatically (without being specified there).

It's just another config package.

So the real list of configs is

```
configs-base/pre,${SKIFF_CONFIG},overrides,configs-base/post
```

peterwilli, 2022-07-21:
Ah, so you think others need this too, so you added it to the default. Anyone who doesn't want it can always do `CONFIG_XFS_FS=n` in their overrides. It's clear now, thanks for explaining.

Image is built so here we go.

paralin, 2022-07-21:
@peterwilli Yes. And while setting `=n` does work, it's more "correct" to unset:

```
# CONFIG_XFS_FS is not set
```

peterwilli, 2022-07-21:
> @peterwilli Yes. And while setting `=n` does work, it's more "correct" to unset:
> 
> ```
> # CONFIG_XFS_FS is not set
> ```

Wow, that's tricky! A comment would be interpreted as something :thinking:

peterwilli, 2022-07-21:
```
root@skiff-0e83fcfb:~# mount /dev/sda1 mnt
root@skiff-0e83fcfb:~# ls mnt
Research  Joplin  phone
```

It works like a charm, thanks so much! Guess it's (finally!) moving day today!

closed 2022-07-21 by peterwilli

paralin, 2022-07-21:
It's `Kconfig` syntax - which was originally developed for the Linux kernel but is also used to configure Buildroot and u-boot.

https://www.openembedded.org/pipermail/openembedded-core/2016-May/120915.html

peterwilli, 2022-07-21:
> It's `Kconfig` syntax - which was originally developed for the Linux kernel but is also used to configure Buildroot and u-boot.
> 
> https://www.openembedded.org/pipermail/openembedded-core/2016-May/120915.html

Thanks for explaining, will read more into that.