skiffos/SkiffOS #220: Question: Why does my device show 4GB of RAM when I really have 8GB?

peterwilli, 2022-07-22:
Hi there,

Not really a bug but I'm surprised to only see 4GB of RAM available:
[image]
I have 8GB on the board. `htop` also only shows 4GB available. Is there a reason for this? I'm curious...

paralin, 2022-07-22:
Mine also is 8GB but shows 4gb,

I see a `dmesg` line very early in the boot:

```
[    0.000000] Memory limited to 4096MB
```

@peterwilli Nice catch, there was a kernel parameter which limited the memory to 4096M in the boot script:

https://github.com/skiffos/SkiffOS/commit/ad85077289ad1e5a473ac1208eb9825315afe3cf

If you pull this commit, then "make cmd/odroid/common/install" once more to update it on your sd card, it should remove the limit.

paralin, 2022-07-22:
Confirmed: that fixed it:

[image]

closed 2022-07-22 by paralin

peterwilli, 2022-07-22:
Thanks so much, I'll try right now and super cool that I found something again, and as always checking out what you changed!

paralin, 2022-07-23:
@peterwilli Yup, just a small mistake in the configs, easy to fix!

Glad it's working properly now 👍🏽