skiffos/SkiffOS #225: WSL build has issues with wlsu utility

clayauld, 2022-08-04:
The ```wslu``` utility doesn't seem to work as expected. The ```wslpath``` binary doesn't get installed and the WSL InteropServer appears broken.

[image]

[image]

paralin, 2022-08-04:
For the `wslu` package we just run "make install" and the makefile provided by wslu does the installing.

It looks like wslpath was renamed to wslupath:

[image]

paralin, 2022-08-04:
@clayauld It needed `/run/WSL` mounted into the skiff system, see https://github.com/skiffos/SkiffOS/pull/226

Please run the following:

```
rm -rf ./workspaces/default/build/skiff-init-*
git fetch
git checkout fix-wslu
git submodule update
make configure compile
```

... to force re-building skiff-init-squashfs.

paralin, 2022-08-04:
@clayauld you'll also need to delete these for the Go update:

```
rm -rf ./workspaces/default/build/host-go-1*
rm -rf ./workspaces/default/host/usr/lib/go
```

paralin, 2022-08-04:
@clayauld #226 should have fixed it - feel free to comment here if not & ill reopen.

clayauld, 2022-08-04:
It appears that the fix doesn't properly mount ```/run/WSL -> /skiff-overlays/system/run/WSL``` as the log says...
[image]

Comparing this to another WSL distro...
[image]

paralin, 2022-08-04:
Someone must mount over /run, I guess systemd. Will look into it

paralin, 2022-08-04:
@clayauld Due to /etc/fstab containing a line mentioning /run systemd mounts a tmpfs to /run.

This must be hiding the bind mount made by skiff-init-squashfs.

paralin, 2022-08-04:
@clayauld Let's try mounting the entire /run dir from the WSL environment and disable systemd `run.mount`

I did that on the fix-wslu branch: https://github.com/skiffos/SkiffOS/pull/227

As before we will need to rebuild skiff-init-squashfs:

```
rm -rf ./workspaces/default/build/skiff-init-*
git fetch
git checkout fix-wslu
git reset --hard origin/fix-wslu
git submodule update
make configure compile
```

clayauld, 2022-08-05:
Hmmm.... Still seems broken.

[image]

paralin, 2022-08-05:
@clayauld are you sure you're on the latest fix-wslu branch commit and have recompiled skiff init as described above?

The logs don't mention /run but they should, there's a line in the config saying to mount /run itself and it should at least show up in those logs there.

clayauld, 2022-08-06:
Yep. You were right. 
[image]

/run/WSL seems to be available now.

When I run the wslu commands I still get the error ```wslpath: command not found```

[image]

I was looking at the source code for wslu and it still looks for the binary or script ```wslpath``` in the PATH. Based on another install of WSL I have running (Pengwin Linux), ```wslpath``` is a symlink to ```/init```. I'm not sure if that is common across WSL distros, but I have found stale versions of re-implimentations of wslpath. None of them seem to work as expected, however.

I can do more digging later this weekend and test other WSL distros to see how they deal with this.

paralin, 2022-08-06:
`wslpath` is a symlink to `/init` which means that `/init` as provided by microsoft implements wslpath if it is started with `wslpath` in `$0`

`/init` doesn't exist inside the skiffos chroot but I will try to find a way to get it in there.

paralin, 2022-08-06:
@clayauld please try `fix-wslpath` branch with https://github.com/skiffos/SkiffOS/pull/229

You will need to rebuild skiff-init-squahfs again as described above (just delete it from `build/` before make compile)

What I did was mount the original / to /mnt/wsl and then symlink wslpath to /mnt/wsl/init

clayauld, 2022-08-06:
Okay That seems to have resolved the issues.

paralin, 2022-08-06:
Great! Thanks for testing

PR #226 (2022-08-04): virt/wsl: add /run/WSL to the bind-mount list
Fixes #225

Mounts /run/WSL into the target.
commit 8380fece (2022-08-04): virt/wsl: add /run/WSL to the bind-mount list

Fixes #225
PR #229 (2022-08-06): virt/wsl: add /mnt/wsl and wslpath symlink
Fixes #225
closed 2022-08-06 by paralin