skiffos/SkiffOS #228: Can't Compile Skiffos

Darin755, 2022-08-06:
Hello,
When I try to compile Skiffos I get the following error:
[image]

It seems that it is trying to fetch dropbear from [skiffos/mirror](https://github.com/skiffos/mirror) but when I look at the repo I don't see dropbear at all. Is this a bug or am I missing something? I should be able to just edit the makefile but I don't know the url to the correct repo.

paralin, 2022-08-06:
It tries the mirror after trying the upstream source. 

If you scroll up a bit you'll see a failed request to the dropbear download servers.

Try again now and see if it works? Their servers were probably just down temporarily.

paralin, 2022-08-06:
Regardless I mirrored dropbear. But yes, it doesn't check the mirror unless the download from the actual source fails first.

closed 2022-08-06 by paralin