skiffos/SkiffOS #230: WSL - OCI runtime error

clayauld, 2022-08-08:
I'm now getting an issue with the WSL build. When trying to run docker containers or connect to skiff-core I'm getting and OCI runtime error.

[image]

clayauld, 2022-08-08:
It appears to be an issue with the cgroup, possibly wrong permissions for root, not being a part of the docker group?

paralin, 2022-08-08:
Will look into it, thanks. And definitely, need to get rid of those absurd long file paths.

paralin, 2022-08-08:
To be more specific: the problem is a quirk in WSL2 where nsenter to a particular process was sending it to the / root namespace instead of the correct one.

paralin, 2022-08-09:
@clayauld I restored the old patch I had before for `runc` which avoids the WSL2 quirk where `mount --move` doesn't seem to work properly within the chroot.

https://github.com/skiffos/SkiffOS/pull/231

This is pretty much a hack but should make runc/docker work properly for now. Hopefully eventually I can figure out why the original behavior in runc doesn't work properly and remove this.

As for the really long paths including /mnt/wsl - I have changed it to bind mount the /wslinit file specifically (not the entire /mnt/wsl tree)

You'll need to rebuild `skiff-init-squashfs` and `runc` on the `fix-wsl-runc` branch:

```
git fetch
git checkout fix-wsl-runc
rm -rf ./workspaces/default/build/runc*
rm -rf ./workspaces/default/build/skiff-init-*
make compile
```

clayauld, 2022-08-09:
That has fixed the issue.
[image]

paralin, 2022-08-09:
Nice!

PR #231 (2022-08-09): virt/wsl: restore patch for runc
This patch might fix #230
closed 2022-08-09 by paralin