skiffos/SkiffOS #232: Issue compiling for pinephone

tag4lucy, 2022-08-15:
Hello,
I have been trying to compile skiffos for pinephone and have been following the sets on the github page and I need to trouble shoot this one with my noob skills and can't find anything on the interzones, 
my system is -
Garuda os
Linux Main-PC 5.19.1-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Thu, 11 Aug 2022 15:22:00 +0000 x86_64 GNU/Linux
#1 ZEN SMP PREEMPT_DYNAMIC Thu, 11 Aug 2022 15:22:00 +0000

and the error I'm getting is,
>>> host-systemd 250.4 Building
PATH="/home/thor/Downloads/SkiffOS/workspaces/default/host/bin:/home/thor/Downloads/SkiffOS/workspaces/default/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/var/lib/snapd/snap/bin" PKG_CONFIG="/home/thor/Downloads/SkiffOS/workspaces/default/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/home/thor/Downloads/SkiffOS/workspaces/default/host/lib/pkgconfig:/home/thor/Downloads/SkiffOS/workspaces/default/host/share/pkgconfig" DESTDIR=/home/thor/Downloads/SkiffOS/workspaces/default/host PYTHONNOUSERSITE=y /home/thor/Downloads/SkiffOS/workspaces/default/host/bin/ninja   -C /home/thor/Downloads/SkiffOS/workspaces/default/build/host-systemd-250.4//build
ninja: Entering directory `/home/thor/Downloads/SkiffOS/workspaces/default/build/host-systemd-250.4//build'
[6/659] Compiling C object src/libsystemd/libsystemd_static.a.p/sd-bus_sd-bus.c.o
FAILED: src/libsystemd/libsystemd_static.a.p/sd-bus_sd-bus.c.o 
/home/thor/Downloads/SkiffOS/workspaces/default/host/bin/ccache /usr/bin/gcc -Isrc/libsystemd/libsystemd_static.a.p -Isrc/libsystemd -I../src/libsystemd -Isrc/basic -I../src/basic -Isrc/fundamental -I../src/fundamental -Isrc/systemd -I../src/systemd -I. -I.. -I../src/libsystemd/sd-bus -I../src/libsystemd/sd-device -I../src/libsystemd/sd-event -I../src/libsystemd/sd-hwdb -I../src/libsystemd/sd-id128 -I../src/libsystemd/sd-journal -I../src/libsystemd/sd-netlink -I../src/libsystemd/sd-network -I../src/libsystemd/sd-resolve -I/home/thor/Downloads/SkiffOS/workspaces/default/host/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=gnu99 -O3 -Wno-format-signedness -Wno-missing-field-initializers-Wno-unused-parameter -Wdate-time -Wendif-labels -Werror=format=2 -Werror=implicit-function-declaration -Werror=incompatible-pointer-types -Werror=int-conversion -Werror=overflow -Werror=override-init -Werror=return-type -Werror=shift-count-overflow -Werror=shift-overflow=2 -Werror=undef -Wfloat-equal -Wimplicit-fallthrough=5 -Winit-self -Wlogical-op -Wmissing-include-dirs -Wmissing-noreturn -Wnested-externs -Wold-style-definition -Wpointer-arith -Wredundant-decls -Wshadow -Wstrict-aliasing=2 -Wstrict-prototypes -Wsuggest-attribute=noreturn -Wunused-function -Wwrite-strings -Wno-maybe-uninitialized -Wno-unused-result -Wno-format-overflow -Werror=missing-declarations -Werror=missing-prototypes -fdiagnostics-show-option -fno-common -fno-strict-aliasing -fstack-protector -fstack-protector-strong -fvisibility=hidden --param=ssp-buffer-size=4 -ffunction-sections -fdata-sections -Werror=shadow -include config.h -O2 -fPIC -pthread -fvisibility=default -MD -MQ src/libsystemd/libsystemd_static.a.p/sd-bus_sd-bus.c.o -MF src/libsystemd/libsystemd_static.a.p/sd-bus_sd-bus.c.o.d -o src/libsystemd/libsystemd_static.a.p/sd-bus_sd-bus.c.o -c ../src/libsystemd/sd-bus/sd-bus.c
In file included from ../src/basic/missing_syscall.h:25,
from ../src/libsystemd/sd-bus/sd-bus.c:38:
../src/basic/missing_stat.h:39:8: error: redefinition of 'struct statx_timestamp'
39 | struct statx_timestamp {
|        ^~~~~~~~~~~~~~~
In file included from /usr/include/bits/statx.h:31,
from /usr/include/sys/stat.h:465,
from ../src/libsystemd/sd-bus/sd-bus.c:9:
/usr/include/linux/stat.h:56:8: note: originally defined here
56 | struct statx_timestamp {
|        ^~~~~~~~~~~~~~~
../src/basic/missing_stat.h:45:8: error: redefinition of 'struct statx'
45 | struct statx STATX_DEFINITION;
|        ^~~~~
/usr/include/linux/stat.h:99:8: note: originally defined here
99 | struct statx {
|        ^~~~~
[10/659] Compiling C object src/libsystemd/libsystemd_static.a.p/sd-device_sd-device.c.o
FAILED: src/libsystemd/libsystemd_static.a.p/sd-device_sd-device.c.o 
/home/thor/Downloads/SkiffOS/workspaces/default/host/bin/ccache /usr/bin/gcc -Isrc/libsystemd/libsystemd_static.a.p -Isrc/libsystemd -I../src/libsystemd -Isrc/basic -I../src/basic -Isrc/fundamental -I../src/fundamental -Isrc/systemd -I../src/systemd -I. -I.. -I../src/libsystemd/sd-bus -I../src/libsystemd/sd-device -I../src/libsystemd/sd-event -I../src/libsystemd/sd-hwdb -I../src/libsystemd/sd-id128 -I../src/libsystemd/sd-journal -I../src/libsystemd/sd-netlink -I../src/libsystemd/sd-network -I../src/libsystemd/sd-resolve -I/home/thor/Downloads/SkiffOS/workspaces/default/host/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=gnu99 -O3 -Wno-format-signedness -Wno-missing-field-initializers-Wno-unused-parameter -Wdate-time -Wendif-labels -Werror=format=2 -Werror=implicit-function-declaration -Werror=incompatible-pointer-types -Werror=int-conversion -Werror=overflow -Werror=override-init -Werror=return-type -Werror=shift-count-overflow -Werror=shift-overflow=2 -Werror=undef -Wfloat-equal -Wimplicit-fallthrough=5 -Winit-self -Wlogical-op -Wmissing-include-dirs -Wmissing-noreturn -Wnested-externs -Wold-style-definition -Wpointer-arith -Wredundant-decls -Wshadow -Wstrict-aliasing=2 -Wstrict-prototypes -Wsuggest-attribute=noreturn -Wunused-function -Wwrite-strings -Wno-maybe-uninitialized -Wno-unused-result -Wno-format-overflow -Werror=missing-declarations -Werror=missing-prototypes -fdiagnostics-show-option -fno-common -fno-strict-aliasing -fstack-protector -fstack-protector-strong -fvisibility=hidden --param=ssp-buffer-size=4 -ffunction-sections -fdata-sections -Werror=shadow -include config.h -O2 -fPIC -pthread -fvisibility=default -MD -MQ src/libsystemd/libsystemd_static.a.p/sd-device_sd-device.c.o -MF src/libsystemd/libsystemd_static.a.p/sd-device_sd-device.c.o.d -o src/libsystemd/libsystemd_static.a.p/sd-device_sd-device.c.o -c ../src/libsystemd/sd-device/sd-device.c
In file included from ../src/basic/stat-util.h:13,
from ../src/basic/chase-symlinks.h:7,
from ../src/libsystemd/sd-device/sd-device.c:11:
../src/basic/missing_stat.h:39:8: error: redefinition of 'struct statx_timestamp'
39 | struct statx_timestamp {
|        ^~~~~~~~~~~~~~~
In file included from /usr/include/bits/statx.h:31,
from /usr/include/sys/stat.h:465,
from ../src/systemd/sd-device.h:22,
from ../src/libsystemd/sd-device/sd-device.c:8:
/usr/include/linux/stat.h:56:8: note: originally defined here
56 | struct statx_timestamp {
|        ^~~~~~~~~~~~~~~
../src/basic/missing_stat.h:45:8: error: redefinition of 'struct statx'
45 | struct statx STATX_DEFINITION;
|        ^~~~~
/usr/include/linux/stat.h:99:8: note: originally defined here
99 | struct statx {
|        ^~~~~
[19/659] Compiling C object src/libsystemd/libsystemd_static.a.p/sd-bus_bus-message.c.o
ninja: build stopped: subcommand failed.
make[3]: *** [package/pkg-generic.mk:293: /home/thor/Downloads/SkiffOS/workspaces/default/build/host-systemd-250.4/.stamp_built] Error 1
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/thor/Downloads/SkiffOS/workspaces/default'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/thor/Downloads/SkiffOS/build'
make: *** [Makefile:2: compile] Error 2

I'm not sure if any more information is required.

paralin, 2022-08-15:
Hi, this is an issue building `host-systemd` which I suppose means there's a duplicate definition of `statx` to your headers at /usr/include

I've updated systemd to v251, the Pinephone kernel to 5.19.1, and Buildroot to latest.

Please rebuild and see if your issue is fixed:

```
make clean
make compile
```

Please select "y" for both options in "make clean"

tag4lucy, 2022-08-16:
Thank you for the fast reply, I gave it a go by dooing git pull them your instructions and kept getting the error, I jumped on my laptop which has the latest Fedora with the following details
Linux 2403-580e-2fd8--9f04-d6df-51a-9ee3.ip6.aussiebb.net 5.18.16-200.fc36.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Aug 3 15:44:49 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
on it and I followed you instructions. This time I got this far,
Checking if your kit is complete...
Looks good
Warning: prerequisite LWP::UserAgent 0 not found.
Warning: prerequisite Test::More 0 not found.
ERROR from evaluation of /home/thor/Downloads/SkiffOS/workspaces/default/build/host-libxml-parser-perl-2.46/Expat/Makefile.PL: Can't locate English.pm in @INC (you may need to install the English module) (@INC contains: /home/thor/Downloads/SkiffOS/workspaces/default/build/host-libxml-parser-perl-2.46/inc /home/thor/Downloads/SkiffOS/workspaces/default/host/lib/perl /usr/local/lib64/perl5/5.34 /usr/local/share/perl5/5.34 /usr/lib64/perl5/vendor_perl /usr/share/perl5/vendor_perl /usr/lib64/perl5 /usr/share/perl5 .) at ./Makefile.PL line 3.
BEGIN failed--compilation aborted at ./Makefile.PL line 3.
make[3]: *** [package/pkg-generic.mk:283: /home/thor/Downloads/SkiffOS/workspaces/default/build/host-libxml-parser-perl-2.46/.stamp_configured] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/thor/Downloads/SkiffOS/workspaces/default'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/thor/Downloads/SkiffOS/build'
make: *** [Makefile:2: compile] Error 2
Is there more info I can give

paralin, 2022-08-16:
I think you might be missing some perl dependencies?

Based on a Google of that error: sudo dnf install perl-core

Does this fix it?

paralin, 2022-08-16:
For reference: https://buildroot.org/downloads/manual/manual.html#requirement

Dependencies:

```
    which
    sed
    make (version 3.81 or any later)
    binutils
    build-essential (only for Debian based systems)
    diffutils
    gcc (version 4.8 or any later)
    g++ (version 4.8 or any later)
    bash
    patch
    gzip
    bzip2
    perl (version 5.8.7 or any later)
    tar
    cpio
    unzip
    rsync
    file (must be in /usr/bin/file)
    bc 
```

You probably already have most of this, except for perl-core

tag4lucy, 2022-08-17:
Thank you for all the help so far, I was able to compile successfully but when I go to format the build onto an sd card it couldn't find the u-boot file.
[thor@2403-580e-2fd8--9f04-d6df-51a-9ee3 SkiffOS]$ sudo bash
[sudo] password for thor:
[root@2403-580e-2fd8--9f04-d6df-51a-9ee3 SkiffOS]# export SKIFF_WORKSPACE=myworkspace
[root@2403-580e-2fd8--9f04-d6df-51a-9ee3 SkiffOS]# export PINE64_SD=/dev/sdc
[root@2403-580e-2fd8--9f04-d6df-51a-9ee3 SkiffOS]# make cmd/pine64/common/format
make[1]: Entering directory '/home/thor/Downloads/SkiffOS/build'
make[2]: Entering directory '/home/thor/Downloads/SkiffOS/configs/pine64/common/extensions'
can't find u-boot image at /home/thor/Downloads/SkiffOS/workspaces/myworkspace/output/images/u-boot.itb
make[2]: *** [Makefile:2: format] Error 1
make[2]: Leaving directory '/home/thor/Downloads/SkiffOS/configs/pine64/common/extensions'
make[1]: *** [Makefile:10: cmd/pine64/common/format] Error 2
make[1]: Leaving directory '/home/thor/Downloads/SkiffOS/build'
make: *** [Makefile:2: cmd/pine64/common/format] Error 2
[root@2403-580e-2fd8--9f04-d6df-51a-9ee3 SkiffOS]#

paralin, 2022-08-18:
@tag4lucy It looks like your compilation that succeeded was with `SKIFF_WORKSPACE=default` (which is the value if SKIFF_WORKSPACE is unset).

But when you tried to install you used `SKIFF_WORKSPACE=myworkspace`.

Please try again but this time don't set `SKIFF_WORKSPACE` (the default is `default`).

tag4lucy, 2022-08-18:
thank you for the help i was able to format and install to an sd card. Something I probably should have said in the beginning is I have a pinephone pro hoping this would work and it not booting, so i will see if any other os is working

paralin, 2022-08-18:
@tag4lucy Pinephone Pro is a bit different, I think you're stuck with whatever bootloader is on emmc.

paralin, 2022-08-18:
@tag4lucy I added a branch `pinephone-pro` with a config `pine64/phone_pro` - could you try that one out?

https://github.com/skiffos/SkiffOS/pull/234

You probably will need to rebuild by deleting `workspaces/default` before running `make compile` again.

tag4lucy, 2022-08-18:
I gave it a go and got this message
[thor@2403-580e-2fd8--9f04-d6df-51a-9ee3 SkiffOS]$ export SKIFF_CONFIG=pine64/phone_pro,core/pinephone_manjaro_kde
[thor@2403-580e-2fd8--9f04-d6df-51a-9ee3 SkiffOS]$ make configure
 ! [pine64/phone_pro] Unknown path! SKIFF_CONFIG_PATH_PINE64_PHONE_PRO not set.
make: *** [Makefile:2: configure] Error 1
[thor@2403-580e-2fd8--9f04-d6df-51a-9ee3 SkiffOS]$
I tried deleting the folder and doing a new git clone and git pull and I cant seem to find the path

paralin, 2022-08-18:
@tag4lucy did you `git checkout pinephone-pro` ?

tag4lucy, 2022-08-18:
I'm getting a few 404 errors
Resolving github.com (github.com)... 20.248.137.48
Connecting to github.com (github.com)|20.248.137.48|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-08-18 12:23:09 ERROR 404: Not Found.

wget --passive-ftp -nd -t 3 -O '/home/thor/Downloads/SkiffOS/workspaces/default/build/.lzo-2.10.tar.gz.vZg84v/output' 'https://github.com/skiffos/mirror/raw/master/lzo-2.10.tar.gz' 
--2022-08-18 12:23:09--  https://github.com/skiffos/mirror/raw/master/lzo-2.10.tar.gz
Resolving github.com (github.com)... 20.248.137.48
Connecting to github.com (github.com)|20.248.137.48|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2022-08-18 12:23:10 ERROR 404: Not Found.

paralin, 2022-08-18:
@tag4lucy please try again, the lzo download servers were temporarily unreachable

tag4lucy, 2022-08-18:
compiling now

tag4lucy, 2022-08-18:
The compiling has finished and took a couple of hours. Now I'm trying to flash using the export SKIFF_WORKSPACE=default
 and getting this error

[root@2403-580e-2fd8--9f04-d6df-51a-9ee3 SkiffOS]# make cmd/pine64/common/format
make[1]: Entering directory '/home/thor/Downloads/SkiffOS/build'
make[2]: Entering directory '/home/thor/Downloads/SkiffOS/configs/pine64/common/extensions'
Are you sure? This will completely destroy all data. [y/N] y
Verify that '/dev/sdc' is the correct device. Be sure. [y/N] y
+ set -e
+ echo 'Formatting device...'
Formatting device...
+ dd if=/dev/zero of=/dev/sdc bs=8k count=13 oflag=dsync
13+0 records in
13+0 records out
106496 bytes (106 kB, 104 KiB) copied, 0.0738055 s, 1.4 MB/s
+ echo 'Creating partitions...'
Creating partitions...
+ sudo partprobe /dev/sdc
Error: Partition(s) 1 on /dev/sdc have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
+ true
+ sudo parted /dev/sdc mklabel msdos
Error: Partition(s) 1 on /dev/sdc have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old
partition(s) will remain in use.  You should reboot now before making further changes.
Ignore/Cancel?

paralin, 2022-08-18:
I think the issue is that the pinephone is booting from the uboot on the emmc which probably isn't recognizing the boot.txt skiff uses. I don't have a pinephone to test right now so trying to figure out what is broken is really a guessing game.

I've read that tow-boot might be needed to be installed to the SPI

Ive also read that in general the pinephone pro way messed up the ability to just boot a SD card without extra work by changing how uboot is read. The regular pinephone was able to boot uboot from a SD card.

paralin, 2022-08-19:
@tag4lucy Sorry to ask you this (I usually have a device and can debug on this end) but do you by any chance have a serial debug monitor that you could use to monitor the boot process?

https://www.youtube.com/watch?v=f5R_hKgDFk8

tag4lucy, 2022-08-19:
I have garbed your email of your profile to communicate with you and I will do everything within my scope to help

paralin, 2022-08-20:
Great sounds good, thanks!

paralin, 2022-09-14:
Update on this: the Pinephone Pro currently makes it past U-boot but the kernel freezes / doesn't output anything. Most likely the load addresses are wrong, and/or the kernel is incompatible somehow. Looking into it further

(thanks @tag4lucy for debugging)

closed 2023-04-17 by paralin