skiffos/SkiffOS #233: Is it normal that I can't boot from USB using intel/x64?

peterwilli, 2022-08-15:
Hey there, it's me again.

paralin, 2022-08-15:
@peterwilli Hey, I haven't tested intel/x64 in a while.

That said, it doesn't install any bootloader, so without some other kind of bootloader it won't boot properly.

Regardless, the partition layout used there is outdated (not the same as the one I'm using on my devices) anyway.

So those two scripts for intel/x64 need to be rewritten.

**Workaround**: for now, please follow these steps: 

 1. Using gparted: clear partition table with gpt partition type.
 2. Create a fat16 partition labeled BOOT with size 300MB.
 3. Create a ext4 partition labeled SKIFFOS with the rest of the disk.
 4. Extract this (Refind) to the EFI partition with `tar -xf efi.tar.gz`: https://drive.google.com/file/d/1aGSzp4aiS6PmkWLHIch3WqVEtKx6_yXb/view
 5. Install to the `SKIFFOS` partition with below commands:

```
export SKIFF_WORKSPACE=laptop
export SKIFF_PARTITION=/dev/disk/by-label/SKIFFOS
make cmd/apple/macbook/install
```

... yes, you're not using a macbook, but that script will install it correctly to the ext4 partition.

This should get you a working bootable USB disk since your bootloader will recognize Refind as EFI on the first partition.

This is also how I boot my desktop & laptop (intel machines).

peterwilli, 2022-08-16:
> ... yes, you're not using a macbook, but that script will install it correctly to the ext4 partition.

😂 

I love the out of the box thinking here.

Could I update it? Sounds like a good PR for me to do it one I get it to boot.

peterwilli, 2022-08-16:
I sadly couldn't run it after following the instructions. Could it be because I have an AMD CPU? I sadly have no PC with Intel lying around to test it out.

Gparted after the steps:
[image]

boot partition tree:
```
/run/media/peter/BOOT
├── EFI
│   └── refind
│       ├── BOOT.CSV
│       ├── drivers_x64
│       ├── icons
│       ├── keys
│       ├── refind.conf
│       └── refind_x64.efi
└── startup.nsh
```

paralin, 2022-08-16:
AMD and Intel are compatible

Hmm could you try editing the BOOT partition and enabling the boot flag?

Also: actually on my desktop machine, I'm using Grub to start Refind and then Refind boots skiff. (a bit convoluted I know). But it should be possible to efi boot straight to refind.

https://apple.stackexchange.com/a/404877

This is a guide on how to make a Refind USB stick, if you follow those instructions, does it boot to the refind menu at least?

If that doesn't work, can you try installing Grub to the USB stick?

If you get it to either the refind or grub menu I can give you instructions on how to add the skiff partition and complete the setup. Just trying to figure out how to get your machine to recognize it as bootable.

peterwilli, 2022-08-17:
Hey there,

Flashing the image did make me able to boot in Refind. I simply did `sudo dd if=refind-flashdrive-0.12.0.img of=/dev/sdb` where `/dev/sdb` is my flash drive. It shows the boot menu. I ended up with the following partitions:

[image]

I'll now try to see if I can add SkiffOS to it using the previous instructions, maybe it works now 🤔

paralin, 2022-08-17:
I think the only difference here is the "boot" flag is set on that first partition.

peterwilli, 2022-08-17:
> I think the only difference here is the "boot" flag is set on that first partition.
> 
> Maybe also the esp flag and remove the msftdata flag

I remember trying it, but not getting it working after all. However, I also found out that one of my USB ports was loose, so it could be it worked all this time, but I randomly didn't get a good connection, so I went back and tried something else while the former would've work (my apologies if that's the case).

peterwilli, 2022-08-17:
[image]

I managed to boot from Refind and select the SkiffOS bzimage (from the 'SKIFFOS' partition I made after flashing Refind on it and using the mac install script) but it got stuck at this screen, seemingly not reading from the drive anymore.

paralin, 2022-08-17:
@peterwilli can you try using `intel/desktop` in your config? It has the settings to use the disk labeled SKIFFOS.

You don't need to rebuild, just set skiff config and make compile, then install to the SKIFFOS partition again. 

Also; what graphics are you using?

peterwilli, 2022-08-17:
What do you mean by what graphics am I using?

Also, it didn't work sadly, I redid make compile using the following settings: `export SKIFF_CONFIG=intel/x64,intel/desktop,skiff/core`

I'll make sure to rebuild just in case, it's not the first time that made things work after all!

paralin, 2022-08-17:
What type of graphics card? Nvidia, or something else?

peterwilli, 2022-08-17:
> What type of graphics card? Nvidia, or something else?

My laptop has a RTX3080 and I think the desktop PC has an integrated GPU as there's no dGPU.

paralin, 2022-08-17:
If you check the SKIFFOS partition, is there any skiff/journal dir on it? If so, can you send?

I'm pretty sure it's not stuck but rather the graphics driver is missing.

peterwilli, 2022-08-17:
[image]
I found these inside the root folder on SKIFFOS partition

paralin, 2022-08-17:
I mean - can you please send me a copy of the `journal` directory inside the `skiff` dir?

`tar -cvzf ./logs.tar.gz ./skiff/journal`

Thanks :)

peterwilli, 2022-08-18:
> I mean - can you please send me a copy of the `journal` directory inside the `skiff` dir?
> 
> `tar -cvzf ./logs.tar.gz ./skiff/journal`
> 
> Thanks :)

Oops, sorry! Here it is: [logs.tar.gz](https://github.com/skiffos/SkiffOS/files/9372256/logs.tar.gz)

paralin, 2022-08-22:
@peterwilli sorry - just remembered to get around to this, will have a look this evening.

paralin, 2022-08-22:
@peterwilli Added nvidia driver (novueau), radeon & amd drivers: f4ace1a6e5b4c604f7bfef32cfd5e8aee32a72f4

To apply the changes to your current build (without rebuilding):

```
git checkout master
git pull
git submodule update
make compile
```

... Buildroot should automatically re-configure linux & add the new modules.

peterwilli, 2022-08-22:
> @peterwilli Added nvidia driver (novueau), radeon & amd drivers: [62915d7](https://github.com/skiffos/SkiffOS/commit/62915d797041a78eadb6c478dbf2ac5f37c8f7bb)
> 
> To apply the changes to your current build (without rebuilding):
> 
> ```
> git checkout master
> git pull
> git submodule update
> make compile
> ```
> 
> ... Buildroot should automatically re-configure linux & add the new modules.

You really did all that? Holy sh*t idk how to thank you, you go to great lengths to support even the most obscure setups! I'll try it this evening.

paralin, 2022-08-23:
@peterwilli No problem!

paralin, 2022-09-02:
Completed adding the new format/install scripts:

 - Added refind to buildroot which compiles rEFInd from source
 - Added format + install scripts for intel/desktop
 - Added more kernel modules to support more hardware

Updated docs: https://github.com/skiffos/SkiffOS/tree/833a097/configs/intel/desktop

This makes a bootable USB stick.

closed 2022-09-02 by paralin

peterwilli, 2022-09-03:
Sorry for not trying yet. I was really busy. I'll start a build right now with the new source ode and see how it goes

commit 833a0974 (2022-09-02): intel/desktop: use refind and implement format/install

Fixes #233