skiffos/SkiffOS #237: Package requeest: Add support for toolbox and distrobox

VarLad, 2022-08-22:
Now that https://github.com/skiffos/SkiffOS/pull/235 is merged, it would be nice to have support for [toolbox](https://github.com/containers/toolbox) and [distrobox](https://github.com/89luca89/distrobox) in SkiffOS.

paralin, 2022-08-23:
Container-toolbox package done: https://github.com/skiffos/buildroot/blob/4fdf3e80f3a04be8dfeea55ce8b030442b34b81a/package/containers-toolbox/containers-toolbox.mk

paralin, 2022-08-23:
Containers toolbox is now available as `apps/toolbx` https://github.com/skiffos/SkiffOS/tree/master/configs/apps/toolbx

paralin, 2022-08-23:
Distrobox package done: https://github.com/skiffos/buildroot/commit/0dc6e8f36975b4f09352da5e40f2daaad0fbd169

paralin, 2022-08-23:
Distrobox added as `apps/distrobox`: https://github.com/skiffos/SkiffOS/tree/master/configs/apps/distrobox

And all done! :)

closed 2022-08-23 by paralin