skiffos/SkiffOS #240: VisionFive Build errors

clayauld, 2022-08-29:
I'm getting build errors again after some of the latest updates.

```
arch/riscv/cpu/mtrap.S: Assembler messages:
arch/riscv/cpu/mtrap.S:65: Error: unrecognized opcode `csrr a0,scause'
arch/riscv/cpu/mtrap.S:66: Error: unrecognized opcode `csrr a1,sepc'
arch/riscv/cpu/mtrap.S:67: Error: unrecognized opcode `csrr a2,stval'
arch/riscv/cpu/mtrap.S:70: Error: unrecognized opcode `csrw sepc,a0'
make[5]: *** [scripts/Makefile.build:301: arch/riscv/cpu/mtrap.o] Error 1
make[5]: *** Waiting for unfinished jobs....
  CC      cmd/riscv/cache_debug.o
arch/riscv/cpu/start.S: Assembler messages:
arch/riscv/cpu/start.S:65: Error: unrecognized opcode `csrw stvec,t0'
arch/riscv/cpu/start.S:72: Error: unrecognized opcode `csrw sie,zero'
arch/riscv/cpu/start.S:347: Error: unrecognized opcode `csrw stvec,t0'
make[5]: *** [scripts/Makefile.build:301: arch/riscv/cpu/start.o] Error 1
  CC      cmd/boot.o
  CC      cmd/bootm.o
  CC      cmd/help.o
  LD      cmd/riscv/built-in.o
arch/riscv/cpu/cpu.c: Assembler messages:
arch/riscv/cpu/cpu.c:96: Error: unrecognized opcode `csrs sstatus,a5'
arch/riscv/cpu/cpu.c:97: Error: unrecognized opcode `csrw 0x003,0'
make[5]: *** [scripts/Makefile.build:266: arch/riscv/cpu/cpu.o] Error 1
make[4]: *** [Makefile:1730: arch/riscv/cpu] Error 2
  CC      cmd/panic.o
make[4]: *** Waiting for unfinished jobs....
  CC      board/starfive/vic7100/vic_ptc.o
  CC      arch/riscv/lib/image.o
  LD      arch/riscv/cpu/fu540/built-in.o
  CC      arch/riscv/lib/boot.o
  CC      cmd/version.o
  CC      arch/riscv/lib/cache.o
  CC      cmd/blk_common.o
  CC      arch/riscv/lib/sbi.o
arch/riscv/lib/cache.c: Assembler messages:
arch/riscv/lib/cache.c:12: Error: unrecognized opcode `fence.i'
make[5]: *** [scripts/Makefile.build:265: arch/riscv/lib/cache.o] Error 1
make[5]: *** Waiting for unfinished jobs....
  CC      cmd/source.o
  CC      cmd/bdinfo.o
  CC      cmd/blkcache.o
  CC      cmd/bootefi.o
CC      cmd/bootz.o
make[4]: *** [Makefile:1730: arch/riscv/lib] Error 2
  CC      cmd/booti.o
  CC      cmd/clk.o
  GZIP    cmd/config_data.gz
  UPD     cmd/config_data_size.h
  CC      cmd/console.o
  CC      cmd/cpu.o
  CC      cmd/echo.o
  CC      cmd/elf.o
  CC      cmd/exit.o
  CC      cmd/ext4.o
board/starfive/vic7100/vic_ptc.c: In function ‘ptc_pinmux_init’:
board/starfive/vic7100/vic_ptc.c:140:25: warning: this ‘for’ clause does not guard... [-Wmisleading-indentation]
  140 |                         for(i=0; i<100; i++) ;
      |                         ^~~
In file included from board/starfive/vic7100/vic_ptc.c:7:
./arch/riscv/include/asm/arch/ezGPIO_fullMux_ctrl_macro.h:13466:30: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘for’
13466 | #define SET_GPIO_7_dout_HIGH { \
      |                              ^
board/starfive/vic7100/vic_ptc.c:141:25: note: in expansion of macro ‘SET_GPIO_7_dout_HIGH’
  141 |                         SET_GPIO_7_dout_HIGH; 
      |                         ^~~~~~~~~~~~~~~~~~~~
  CC      cmd/ext2.o
  CC      cmd/fat.o
  CC      cmd/fdt.o
  CC      cmd/fs.o
  CC      cmd/itest.o
  GZIP    cmd/license_data.gz
  CC      cmd/load.o
  UPD     cmd/license_data_size.h
  CC      cmd/log.o
  CC      cmd/lsblk.o
CC      cmd/mem.o
  CC      cmd/sleep.o
  CC      cmd/mmc.o
  CC      cmd/legacy-mtd-utils.o
  CC      cmd/net.o
  CC      cmd/part.o
  CC      cmd/pxe.o
  CC      cmd/pxe_utils.o
  CC      cmd/sf.o
  CC      cmd/setexpr.o
  CC      cmd/strings.o
  CC      cmd/sysboot.o
  CC      cmd/test.o
  CC      cmd/unlz4.o
  CC      cmd/fs_uuid.o
  CC      cmd/ximg.o
  CC      cmd/gpt.o
  CC      cmd/nvedit.o
  UPD     cmd/license_data_gz.h
  CC      cmd/license.o
  UPD     cmd/config_data_gz.h
  CC      cmd/config.o
board/starfive/vic7100/vic7100.c: In function ‘board_init’:
board/starfive/vic7100/vic7100.c:1153:15: warning: implicit declaration of function ‘cache_enable_ways’ [-Wimplicit-function-declaration]
 1153 |         ret = cache_enable_ways();
      |               ^~~~~~~~~~~~~~~~~
  LD      board/starfive/vic7100/built-in.o
  LD      cmd/built-in.o
make[3]: *** [package/pkg-generic.mk:293: /home/clayauld/SkiffOS/workspaces/visionfive/build/uboot-custom/.stamp_built] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/clayauld/SkiffOS/workspaces/visionfive'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/clayauld/SkiffOS/build'
make: *** [Makefile:2: compile] Error 2

```

paralin, 2022-08-30:
https://www.mail-archive.com/u-boot@lists.denx.de/msg430287.html

I found the relevant bug, I think. Will post a fix here shortly after reproducing the issue.

paralin, 2022-08-30:
https://github.com/skiffos/SkiffOS/commit/09df241c7d5c02fb95650d8c4333ae3097a5ac61

Fixed with this commit.

This was the patch added:

 - https://lore.kernel.org/all/PH7PR14MB5594FD11D1BE74284F554BEBCED49@PH7PR14MB5594.namprd14.prod.outlook.com/

There is another relevant one (but superseded by the one I applied):

 - https://lore.kernel.org/all/Yh2RTbmwDJIg+R0j@ubuntu01/T/#r3ee691a1967a027f9e33736f5e40da1effd201cb

The patch in the u-boot tree is here:

 - https://github.com/skiffos/u-boot/commit/3bb93519563146419aa06e8d21394089163302b6

Thanks for the report!

closed 2022-08-30 by paralin