skiffos/SkiffOS #242: starfive/visionfive: riscv assembler errors building opensbi

clayauld, 2022-09-01:
We're now getting compile errors building opensbi for the Starfive Visionfive:

```
/home/clayauld/SkiffOS-ext/skiffos/workspaces/visionfive/build/opensbi-custom/lib/sbi/sbi_tlb.c: Assembler messages:
/home/clayauld/SkiffOS-ext/skiffos/workspaces/visionfive/build/opensbi-custom/lib/sbi/sbi_tlb.c:175: Error: unrecognized opcode `fence.i'
 CC        lib/utils/serial/fdt_serial_sifive.o
make[5]: *** [Makefile:339: /home/clayauld/SkiffOS-ext/skiffos/workspaces/visionfive/build/opensbi-custom/build/lib/sbi/sbi_tlb.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [package/pkg-generic.mk:293: /home/clayauld/SkiffOS-ext/skiffos/workspaces/visionfive/build/opensbi-custom/.stamp_built] Error 2
make[3]: *** [Makefile:23: _all] Error 2
make[3]: Leaving directory '/home/clayauld/SkiffOS-ext/skiffos/workspaces/visionfive'
make[2]: *** [Makefile:64: compile] Error 2
make[2]: Leaving directory '/home/clayauld/SkiffOS-ext/skiffos/build'
make[1]: *** [Makefile:2: compile] Error 2
make[1]: Leaving directory '/home/clayauld/SkiffOS-ext/skiffos'
make: *** [Makefile:7: compile] Error 2
```

paralin, 2022-09-02:
Another unrecognized opcode, I think it just needs a patch / update. Will do shortly.

paralin, 2022-09-02:
@clayauld I updated opensbi - could you please `git pull` then `git submodule update` then `make br/opensbi-dirclean` then `make compile` ?

Hopefully fixes it.

clayauld, 2022-09-02:
Yes. That fixes the build issue.

closed 2022-09-02 by paralin