skiffos/SkiffOS #244: intel/desktop: use GRUB2 to chainload rEFInd

paralin, 2022-09-17:
Some platforms (Lenovo Laptops) don't detect the rEFInd EFI correctly.

Use grub2 to chainload the rEFInd binary:

```
set default="0"
set timeout="1"

menuentry "Refind" {
        chainloader /EFI/refind/refind_x64.efi
}
```

paralin, 2022-09-17:
cc @peterwilli not sure if you have done your testing w/ intel/desktop yet but this is relevant.

peterwilli, 2022-09-19:
I have but not the second distro thing yet. I'll try both today (I hope, have to prepare a lot for vacation)

paralin, 2022-10-12:
Most EFi issues are solved with fixes to refind, there were some mistakes in how it was setup. I'll re-open this again if I can't figure out how to get refind to work reliably.

closed 2022-10-12 by paralin