skiffos/SkiffOS #246: WSL: Recent additions of rEFInd breaks build

clayauld, 2022-10-04:
Recent additions of the rEFInd bootloader for x64 breaks the WSL build:

```
chmod a-x btrfs_x64.efi
mkdir -p ../drivers_x64
cp btrfs_x64.efi ../drivers_x64
>>> refind 0.13.3.1 Installing to target
>>> refind 0.13.3.1 Installing to images directory
if [ -d /debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/images/efi-part/EFI/refind ]; then rm -rf /debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/images/efi-part/EFI/refind; fi
cd /debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/build/refind-0.13.3.1 && InstallDir="/debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/images/efi-part" OSTYPE="linux" Platform="x64" RootDir="/debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/images/efi-part" ./refind-install --yes --alldrivers
The --alldrivers option is meant for creating USB flash drives with (near-)
universal boot support. Using it on a hard disk partition runs the risk of
creating serious problems, up to and including rendering your computer
unbootable. Are you SURE you want to proceed (Y/N)? Y
ShimSource is none
Installing rEFInd on Linux....
Running in BIOS mode with no existing default boot loader; installing to
/debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/images/efi-part/EFI/BOOT
Note: IA32 (x86) binary not installed!
Copied rEFInd binary files

Copying sample configuration file as refind.conf; edit this file to configure
rEFInd.

Installation has completed successfully.

if [ -d /debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/images/efi-part/EFI/BOOT ]; then rm -rf /debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/images/efi-part/EFI/BOOT; fi
mv /debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/images/efi-part/EFI/refind/ /debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/images/efi-part/EFI/BOOT/
mv: cannot stat '/debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/images/efi-part/EFI/refind/': No such file or directory
make[4]: *** [package/pkg-generic.mk:375: /debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl/build/refind-0.13.3.1/.stamp_images_installed] Error 1
make[3]: *** [Makefile:23: _all] Error 2
make[3]: Leaving directory '/debian/SkiffOS-ext/skiffos_paralin/workspaces/wsl'
make[2]: *** [Makefile:64: compile] Error 2
make[2]: Leaving directory '/debian/SkiffOS-ext/skiffos_paralin/build'
make[1]: *** [Makefile:2: compile] Error 2
make[1]: Leaving directory '/debian/SkiffOS-ext/skiffos_paralin'
make: *** [Makefile:7: compile] Error 2

```

paralin, 2022-10-04:
rEFInd isn't needed for WSL. I'll disable it.

paralin, 2022-10-04:
Fixed, but you've brought up another issue: rEFInd is apparently checking if the host system has a existing bios bootloader or not, which is obviously wrong. I'll look into that too.

commit a7b7a098 (2022-10-04): virt/wsl: disable refind bootloader

Fixes #246
closed 2022-10-04 by paralin