skiffos/SkiffOS #248: qemu/riscv64: Add support for new Qemu virt

clayauld, 2022-10-07:
I'd like to use Qemu to do some build testing inside SkiffOS. Wondering how much work it would take to add an additional Qemu build for the RISCV64 architecture?

Looks like upstream buildroot already has the kernel defconfig built for two options: one with MMU support and one without
* https://github.com/skiffos/buildroot/blob/aceded7991a489df4e0fb7b336ca133de38245b9/configs/qemu_riscv64_virt_defconfig
* https://github.com/skiffos/buildroot/blob/aceded7991a489df4e0fb7b336ca133de38245b9/configs/qemu_riscv64_nommu_virt_defconfig

An additional point of reference may be the Ubuntu build for Qemu RISCV64: https://cdimage.ubuntu.com/releases/22.04.1/release/ubuntu-22.04.1-preinstalled-server-riscv64+unmatched.img.xz

I've been able to get this running in Qemu using the following Qemu command:

```
qemu-system-riscv64 \
  -machine virt \
  -nographic \
  -m 4G -smp 8 \
  -bios /usr/lib/riscv64-linux-gnu/opensbi/generic/fw_jump.elf \
  -kernel /usr/lib/u-boot/qemu-riscv64_smode/uboot.elf \
  -device virtio-net-device,netdev=eth0 \
  -netdev user,id=eth0,hostfwd=tcp::2222-:22 \
  -drive file=ubuntu-22.04.1-preinstalled-server-riscv64+unmatched.img,format=raw,if=virtio
```

clayauld, 2022-10-07:
Oh! Excellent! Where is this documented? I didn't see it in the configs.

paralin, 2022-10-07:
@clayauld https://github.com/skiffos/SkiffOS#qemu

I've added some more info there about selecting alternative architectures too.

Does this solve the issue for your use cases? It's using opensbi built-in to qemu.

clayauld, 2022-10-07:
Yes it does!

Thank you for the amazingly fast response, as always.

closed 2022-10-07 by clayauld