skiffos/SkiffOS #251: Pi3/NixOS

sstent, 2022-10-24:
SKIFF_CONFIG=pi/3,core/nixos

Boot fails to start NixOS as upstream NixOS image repo doesn't support ARMHF and this seems to bypass the trigger to build the image locally.

paralin, 2022-10-24:
@sstent As a temporary solution you could try enabling arm64 in the config.txt for the pi 3, because pi3 actually does have the ability to run arm64 code

paralin, 2026-10-29:
Fixed on master.

NixOS publishes no 32-bit ARM image, so core/nixos needs an aarch64 or x86_64 target. 218ce9ee made the skiff-core-nixos package depend on one, but on `pi/3` (32-bit) Kconfig then dropped the package without a word and the image built with no NixOS core. 4a23a79e makes `make configure` print every requested option that was dropped, so `pi/3,core/nixos` now warns about `BR2_PACKAGE_SKIFF_CORE_NIXOS`.

The Pi 3 runs 64-bit code, so use the 64-bit config:

```
export SKIFF_CONFIG=pi/3x64,core/nixos
```

closed 2026-10-29 by paralin