skiffos/SkiffOS #252: Jetson Nano build failing

sphw, 2022-10-25:
I am trying to build SkiffOS for the Jetson Nano, with `SKIFF_CONFIG=jetson/nano`. I am getting a consistent build failure with the following output 

```
checking for suffix of object files... configure: error: in `/home/sphw/SkiffOS/workspaces/default/build/glibc-2.36-44-g2628500f5dff1dd99c49a09b418b3b1ea3a6b5d3/build':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make[3]: *** [package/pkg-generic.mk:283: /home/sphw/SkiffOS/workspaces/default/build/glibc-2.36-44-g2628500f5dff1dd99c49a09b418b3b1ea3a6b5d3/.stamp_configured] Error 1
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/home/sphw/SkiffOS/workspaces/default'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/sphw/SkiffOS/build'
make: *** [Makefile:2: compile] Error 2
```

Here is the output of `config.log`. [config.log](https://github.com/skiffos/SkiffOS/files/9857429/config.log)

The relevant error appears to be:
```
cc1: error: bad value 'cortex-a57+crypto' for '-mtune=' switch
cc1: note: valid arguments to '-mtune=' switch are: nocona core2 nehalem corei7 westmere sandybridge corei7-avx ivybridge core-avx-i haswell core-avx2 broadwell skylake skylake-avx512 cannonlake icelake-client rocketlake icelake-server cascadelake tigerlake cooperlake sapphirerapids alderlake bonnell atom silvermont slm goldmont goldmont-plus tremont knl knm intel x86-64 eden-x2 nano nano-1000 nano-2000 nano-3000 nano-x2 eden-x4 nano-x4 k8 k8-sse3 opteron opteron-sse3 athlon64 athlon64-sse3 athlon-fx amdfam10 barcelona bdver1 bdver2 bdver3 bdver4 znver1 znver2 znver3 btver1 btver2 generic
```

Is this a misconfiguration on my end? Or an actual issue with the Jetson Nano config? I've been able to repo on two different machines, but they are both running Debian-based distros with similar envs.

paralin, 2022-10-25:
@sphw This should be fixed now. `git pull && git submodule update` on `master` & it should work-

if not, try deleting `./configs/jetson/nano/cflags` directory entirely & try again + let me know here that you had to do that.

Thanks!

closed 2022-10-25 by paralin

paralin, 2022-10-26:
@sphw I see now that jetson nano is missing a config file... fixed in b8fea928bb73e7e87b886e186bca49fb4fe931d8

Please follow these steps:

```
git pull && git submodule update
# select y for clearing the compiler cache here
make clean
make compile
```

commit f0fa0297 (2022-10-25): jetson/nano: remove incompatible mcpu flag

GCC 12 now no longer supports the mcpu flag format we used before.

Fixes #252