skiffos/SkiffOS #254: Jetson Nano build missing `linux4tegra`

sphw, 2022-10-26:
I'm back with another Jetson Nano build issue. After the latest commits the build completes successfully, but can't be flashed to an SD card due to this error: `linux4tegra tools not found, ensure buildroot is done compiling.`. 

It seems like `linux4tegra` is meant to be downloaded by `linux4tegra-legacy`, but it seems to not happen. Not sure why this is the case.

paralin, 2022-10-27:
@sphw Thanks for the report, indeed, there was a typo, fixed: d0c026e3801ed351dfc18e377bd39e3dc4824957

If you `git pull` then `git submodule update` then `make configure compile` it should compile the missing package and add a symlink at workspaces/default/images/linux4tegra - let me know if this fixes it! Thanks again.

sphw, 2022-10-27:
I cloned the latest repo and rebuilt, but the folder still doesn't seem to get symlinked.

paralin, 2022-10-27:
@sphw Sigh... My bad... I finally was able to run a test build just now and noticed this: 

```
+++ b/configs/jetson/common/buildroot_ext/package/linux4tegra-legacy/Config.in
 config BR2_PACKAGE_LINUX4TEGRA_LEGACY
+	bool "linux4tegra-legacy"
 	help
 	  Supports discontinued boards: Jetson TX2, Nano.
```

Fixed: 0d7fffb45127ab1e03d324b7ae6ffad608a06b91

You shouldn't need to rebuild - just `git pull` then `make compile` again & it will pick up the change.

paralin, 2022-10-28:
It seemed to work correctly in my test build on `master` - let me know if it's not fixed for you!

Thanks again for the report!

closed 2022-10-28 by paralin

sphw, 2022-10-28:
Yup it all worked for me. Thanks for your help!