skiffos/SkiffOS #259: Cannot build on arch linux, kernel 6.09

symgryph, 2022-11-28:
When I attempt to build (after having run a successful compile), I tried 2 things: 1. directly to sd card. Format works fine, but the  install image fails.3. 

2. tried building to image, but I get the following error:

```ini
/home/tmunn/src/SkiffOS/scripts/execute_command.sh: line 35: /home/tmunn/src/SkiffOS/workspaces/.config_odroid//bind_env.sh: No such file or directory
make[2]: Entering directory '/home/tmunn/src/SkiffOS/configs/odroid/common/extensions'
Allocating sparse image...
Setting up loopback device...
Using loopback device at /dev/loop0
Cannot find /home/tmunn/src/SkiffOS/workspaces/odroid/output/images/hk_sd_fuse/sd_fusing.sh, make sure Buildroot is compiled.
Removing loopback device...
make[2]: *** [Makefile:8: buildimage] Error 1
make[2]: Leaving directory '/home/tmunn/src/SkiffOS/configs/odroid/common/extensions'
make[1]: *** [Makefile:10: cmd/odroid/common/buildimage] Error 2
make[1]: Leaving directory '/home/tmunn/src/SkiffOS/build'
make: *** [Makefile:2: cmd/odroid/common/buildimage] Error 2
```
I did set my environment variables according to the directions. Compile completed successfully. When trying to build to sd card (/dev/sdb) I get:

```ini
unmounting /tmp/tmp.BWhaAOVpoQ/boot...
umount: /tmp/tmp.BWhaAOVpoQ/boot: not mounted.
make[2]: *** [Makefile:5: install] Error 32
make[2]: Leaving directory '/home/tmunn/src/SkiffOS/configs/odroid/common/extensions'
make[1]: *** [Makefile:10: cmd/odroid/common/install] Error 2
make[1]: Leaving directory '/home/tmunn/src/SkiffOS/build'
make: *** [Makefile:2: cmd/odroid/common/install] Error 2
```

paralin, 2022-11-28:
@symgryph check the workspaces/ directory, is there one named odroid? The `SKIFF_WORKSPACE` variable should be set to the same value as you used during the compilation process. If you didn't set it before, it will be named "default"

symgryph, 2022-11-28:
trying now

symgryph, 2022-11-28:
```iniexport SKIFF_WORKSPACE=odroid
[root@violet SkiffOS]# make cmd/odroid/common/install
make[1]: Entering directory '/home/tmunn/src/SkiffOS/build'
make[2]: Entering directory '/home/tmunn/src/SkiffOS/configs/odroid/common/extensions'
Using boot.txt at /home/tmunn/src/SkiffOS/configs/odroid/m1/resources/boot-scripts/boot.txt
Using board-specific install_sd.sh
Mounting /dev/sdb1 to /tmp/tmp.dcXcqtKiEi/boot...
mount: /tmp/tmp.dcXcqtKiEi/boot: cannot mount /dev/sdb1 read-only.
       dmesg(1) may have more information after failed mount system call.
Unmounting /tmp/tmp.dcXcqtKiEi/boot...
umount: /tmp/tmp.dcXcqtKiEi/boot: not mounted.
make[2]: *** [Makefile:5: install] Error 32
make[2]: Leaving directory '/home/tmunn/src/SkiffOS/configs/odroid/common/extensions'
make[1]: *** [Makefile:10: cmd/odroid/common/install] Error 2
make[1]: Leaving directory '/home/tmunn/src/SkiffOS/build'
make: *** [Makefile:2: cmd/odroid/common/install] Error 2
```

I did set my workspace to odroid, and the name of the workspace is odroid.

symgryph, 2022-11-28:
here are my variables after a compile (as non root), I am ROOT when running the commands, however, just not during the compile.

```ini
set | grep -i skiff
OLDPWD=/home/tmunn/src/SkiffOS/workspaces
PWD=/home/tmunn/src/SkiffOS
SKIFF_WORKSPACE=odroid
[root@violet SkiffOS]# set | grep -i sdb
ODROID_SD=/dev/sdb
```

paralin, 2022-12-01:
@symgryph Please check `sudo dmesg` logs and see if there are any warnings related to `mount`. it seems like the `mount /dev/sdb1 ./mountpoint` failed for some reason.

paralin, 2022-12-17:
@symgryph I think if you have any issue installing directly to sd card (check dmesg for errors if so), you can use `buildimage` instead as an intermediate step (flashing the image with dd after). Just make sure you have SKIFF_WORKSPACE set properly when you run the buildimage command.

Lmk if you need more help with this!

closed 2022-12-17 by paralin