skiffos/SkiffOS #262: container tweaks needed to run KDE Plasma

bam80, 2022-12-19:
Are similar container tweaks needed as those described for Distrobox?:
https://github.com/89luca89/distrobox/blob/main/docs/posts/run_latest_gnome_kde_on_distrobox.md#running-latest-plasma

If not, why is the difference?

paralin, 2022-12-19:
No. Kde plasma runs fine. I don't know much about distrobox but I would assume the difference is we run the Wayland server inside the container rather than outside.

Feel free to open an issue if you run into any problems!

closed 2022-12-19 by paralin

paralin, 2022-12-20:
For one, SkiffOS runs a systemd instance in the container, so forwarding to the parent instance is unnecessary.

Besides that, yes, a lot of work has been put into making SkiffOS core run any distribution correctly in the container without needing any modifications.

bam80, 2022-12-20:
> No. Kde plasma runs fine.

What container you tried? I didn't find pre-ready Desktop Plasma containers here, so you installed the Plasma packages in one of the existing I guess?

paralin, 2022-12-20:
Yes. Nothing special, all you need to do is apt install it, or whatever package manager. I've tried fedora, debian, and Gentoo with kde.