skiffos/SkiffOS #263: cachyos request

osevan, 2022-12-20:
Could you try arch based cachyos functionality with skiffos? 

Thanks and

Best regards

paralin, 2022-12-22:
I guess you mean adding the patches from here:

 - https://github.com/CachyOS/linux-cachyos

I could possibly add it as a buildroot package which adds the patches to the buildroot linux patches list

osevan, 2023-01-02:
> I guess you mean adding the patches from here:
> 
>     * https://github.com/CachyOS/linux-cachyos
> 
> 
> I could possibly add it as a buildroot package which adds the patches to the buildroot linux patches list

Ehm no, not the patches,  because its kernel patches,  i dont think you could do containerized kernels work, because containers using host kernel.

But, as distribution inside container like gentoo and debian,  you could try to build.

Cachyos is arch based distribution with nice repository. 
When someone want use cachyos repository,  they could install cachyos image, with optimzed pkgs.

paralin, 2023-01-02:
Ok, I understand. It can be added as a core config. However, the patches with the other scheduler (bore) can actually be added to skiffos since SkiffOS compiles its own kernel. (depending on the target system)

paralin, 2023-07-19:
Closing as out-of-scope for now

closed 2023-07-19 by paralin