skiffos/SkiffOS #270: Workspace buildroot config does not override packages

realnedsanders, 2023-02-24:
Using `make br/menuconfig`
saved to default config file `workspaces/$workspace/.config`
does not override any buildroot config options contained in any of the packages.

paralin, 2023-02-24:
@realnedsanders the changes you make with menuconfig are not persisted and will be overridden the next time "make configure" runs.

Instead, add a file at ./overrides/buildroot/myconfig and add the configuration options there. `BR2_PACKAGE_NANO=y` for example.

Then run "make configure" and it will apply the overrides.

I'll rename this issue because I've been thinking of a way to detect what was changed in menuconfig and automatically store it under overrides. Please use the approach above in the meantime.

realnedsanders, 2023-02-24:
Ahhh, that makes sense, I feel foolish.
Persisting the menuconfig options to overrides/ would be very convenient.
`diff <(sort workspaces/.config_$workspace/final/buildroot) <(sort workspaces/$workspace/.config)` 
after the user exits `menuconfig` might be useful.

paralin, 2023-02-24:
In the kernel there's a script named diffconfig that generates those diffs.

closed 2026-09-22 by realnedsanders