skiffos/SkiffOS #273: skiff-init-squashfs: resizing persist partition fails

paralin, 2023-02-25:
Let's say that we boot to a disk at /, then resize2fs will happily online resize it. 

But what we do in skiff-init-squashfs is then mount an overlayfs and a squashfs from / and then chroot into a different root. Then what happens is / is now a chroot, resize2fs doesn't detect that it needs online resizing, and fails to resize it.

There is a fix for this somewhere in making a patch for resize2fs to always do online resizing.

Related: #272 

cc @maxberger

maxberger, 2023-02-25:
Sounds fun! I created that patch:

https://github.com/tytso/e2fsprogs/compare/master...maxberger:e2fsprogs:master

It is yet untested, but it's a start.

paralin, 2026-10-29:
Verified the fix from aa040c01 and closing.

To reproduce the early-boot case, I mounted a small ext4 image, chrooted into it with `/dev` bound but no `/proc`, and resized it from inside. Stock resize2fs 1.47.4 refuses: "Can't check if filesystem is mounted due to missing mtab file". With the patch, `resize2fs -F -o /` takes the online path and grows the filesystem from 64 MiB to 256 MiB, and `e2fsck -fn` finds no errors afterwards. skiff-init-squashfs passes `-o /` only when the resize device is the block device mounted at `/`.

closed 2026-10-29 by paralin