skiffos/SkiffOS #274: ${BUILDROOT_DIR}/host/bin/qemu-system is missing

yokoyama-flogics, 2023-03-01:
Hello,

First, thanks for the great project!

I have a problem mentioned in the subject.  It might be a silly mistake, but please help.

I ran the following steps, as mentioned [here](https://github.com/skiffos/SkiffOS/tree/master/configs/virt/qemu). by 2022.11.1 release.  Host OS is Ubuntu 18.04.5 LTS.

```bash
$ export SKIFF_WORKSPACE=qemu
$ export SKIFF_CONFIG=intel/x64,virt/qemu
$ make configure
$ make compile
$ sudo apt install qemu-utils
$ make cmd/virt/qemu/run
```

However, I saw the following error.

```
/home/yokoyama/SkiffOS/configs/virt/qemu/scripts/execute_qemu.sh: line 70:
    /home/yokoyama/SkiffOS/workspaces/qemu/host/bin/qemu-system: No such file or directory
```

I don't see qemu-system in `workspaces/qemu/host/bin`.

What step is required to build the qemu-system?

Any suggestion would be appreciated.

Regards,

Atsushi

paralin, 2023-03-01:
@yokoyama-flogics Thanks for the report.

qemu-system is provided by `BR2_PACKAGE_HOST_QEMU` and the build dir for this should exist under `workspaces/qemu/build/host-qemu-7.2.0`.

Does that directory exist? If not, does it exist after running "make br/host-qemu" in the root of the SkiffOS tree?

I'll run a test build on my end as well.

yokoyama-flogics, 2023-03-01:
@paralin ,

Thanks for the prompt reply!

> Does that directory exist? If not, does it exist after running "make br/host-qemu" in the root of the SkiffOS tree?

It didn't exist, but running `make br/host-qemu`, successfully found the `host-qemu-7.2.0` and the symbolic link `workspaces/qemu/host/bin/qemu-system`.

However, the symbolic link target `qemu-system-x86_64` is still missing.  Even running `make compile` again, it doesn't look generated.

I will check my steps once again.

Thanks in anyway!

Regards,

Atsushi

paralin, 2023-03-01:
@yokoyama-flogics Please cd to the `workspaces/qemu` directory and then run `make menuconfig` then press `/` to search. Search for `BR2_PACKAGE_HOST_QEMU_SYSTEM_MODE` and let me know if that's set to y or not. You can also force a rebuild of that package with `make br/host-qemu-dirclean br/host-qemu`

yokoyama-flogics, 2023-03-01:
Hi @paralin 

`BR2_PACKAGE_HOST_QEMU_SYSTEM_MODE` is set to `n` (no).  I will try `make br/host-qemu-dirclean br/host-qemu`.

Thanks!

paralin, 2023-03-01:
@yokoyama-flogics There must be something wrong with the `make configure` step.

Please cd to the skiffos root dir and then:

```
export SKIFF_WORKSPACE=qemu
export SKIFF_CONFIG=intel/x64,virt/qemu
make configure
```

Then check the menuconfig again and see if `BR2_PACKAGE_HOST_QEMU` is set (and also `_SYSTEM_MODE`).

They are defined here, and if they aren't set after `make configure` then something must be wrong with the configure script: https://github.com/skiffos/SkiffOS/blob/master/configs/virt/qemu/buildroot/tools#L1

yokoyama-flogics, 2023-03-01:
@paralin ,

I ran

```
export SKIFF_WORKSPACE=qemu
export SKIFF_CONFIG=intel/x64,virt/qemu
make configure
```

without any errors, but still `BR2_PACKAGE_HOST_QEMU` and `BR2_PACKAGE_HOST_QEMU_SYSTEM_MODE` are both set to `n` (no).

I will be out of my lab for a while, so the next reply would be delayed. 

Thanks,

Atsushi

paralin, 2023-03-01:
@yokoyama-flogics Okay thanks for testing, I'll have a look and see if I can find the cause.

paralin, 2023-03-01:
@yokoyama-flogics I ran the following steps:

```
git clone https://github.com/skiffos/skiffos
cd skiffos
git checkout 2022.11.1
git submodule update --init
export SKIFF_WORKSPACE=qemu
export SKIFF_CONFIG=intel/x64,virt/qemu
make configure
make br/menuconfig
```

After this, `BR2_PACKAGE_HOST_QEMU` is correctly set to `y` in the config.

Can you try `git submodule update` in the skiffos root? Perhaps you have a different version of buildroot/ checked out.

Thanks again for testing!

yokoyama-flogics, 2023-03-01:
@paralin ,

Thanks for your help and reproduction.

However, `git submodule update --init` didn't help by my side, unfortunately.

After submodule update, I see the following.

```bash
$ cd buildroot
$ git log|head
commit aa65881a01c81a01cb18fef71152050413d23786
Author: Christian Stewart <christian@paral.in>
Date:   Thu Sep 1 22:21:53 2022 -0700

    [WIP] boot/refind: new package
    
    rEFInd is a user-friendly EFI GUI for boot selection.
    
    It can be used as an alternative to Grub with a user-friendly graphical and/or
    text interface for selecting a boot option.
```

Is this an incorrect revision?

With this revision, I ran again

```
export SKIFF_WORKSPACE=qemu
export SKIFF_CONFIG=intel/x64,virt/qemu
make configure
make br/menuconfig
```

but `BR2_PACKAGE_HOST_QEMU` is still `n`...

I will try again from the very beginning in my spare time.

BTW, SkiffOS master branch's buildroot submodule refers `buildroot @ 0e8570e`.
Should we use that revision instead?

Thanks and Regards,

Atsushi

paralin, 2023-03-01:
@yokoyama-flogics I would recommend checking out SkiffOS `master`, then running `git submodule update` - since I will be releasing 2023.02 soon and that branch is stable at the moment.

It's strange that our two machines are not reproducing it the same way. I'm using Gentoo (latest versions of everything) so maybe, somehow,  Ubuntu 18.04.5 LTS. causes the different behavior. But I don't think so - they should be the same.

Very odd that it doesn't correctly apply BR2_PACKAGE_HOST_QEMU.

yokoyama-flogics, 2023-03-01:
@paralin ,

OK.  I will try master branch this evening!

Thank you for your suggestion!!
And I won't forget `git submodule update` in the top directory from now on.

Regards,

Atsushi

yokoyama-flogics, 2023-03-01:
@paralin ,

I found a problem.  That is, default gcc of Ubuntu 18 is version 7.

After reading the discussion below, and I enabled gcc-8.  Now `make br/menuconfig` successfully showed `BR2_PACKAGE_HOST_QEMU` and `BR2_PACKAGE_HOST_QEMU_SYSTEM_MODE` are both `y` (yes).

- [Install gcc-8 only on Ubuntu 18.04? - Ask Ubuntu](https://askubuntu.com/questions/1028601/install-gcc-8-only-on-ubuntu-18-04)

Now I making `compile`.  I will report the result later.

Thanks,

Atsushi

paralin, 2023-03-01:
@yokoyama-flogics Quite odd that the GCC version impacts it, but perhaps this is something in Buildroot.

yokoyama-flogics, 2023-03-01:
@paralin ,

Thank you for your investigation.

I'm not familiar (at all) with Buildroot, I found the following in `buildroot/package/qemu/Config.in.host`.

```
comment "host-qemu needs a host gcc >= 8"
        depends on BR2_PACKAGE_HOST_QEMU_SYSTEM_ARCH_SUPPORTS || \
                BR2_PACKAGE_HOST_QEMU_USER_ARCH_SUPPORT
        depends on !BR2_HOST_GCC_AT_LEAST_8

config BR2_PACKAGE_HOST_QEMU
        bool "host qemu"
        depends on BR2_PACKAGE_HOST_QEMU_SYSTEM_ARCH_SUPPORTS || BR2_PACKAGE_HOST_QEMU_USER_ARCH_SUPPORTS
        depends on BR2_HOST_GCC_AT_LEAST_8
```

The fact is, it is also shown in the `make menuconfig` screen.  But it was hard to find that by searching by the `/` key.

In anyway, I hope I can report the compilation result by tomorrow! :)

Regards,

Atsushi

paralin, 2023-03-01:
@yokoyama-flogics well, there you have it! It requires a newer gcc.

I'll update the docs, thanks for looking into this.

paralin, 2023-03-01:
@yokoyama-flogics feel free to reopen and / or make a new issue if you run into any further difficulty. Thanks again for testing.

closed 2023-03-01 by paralin

yokoyama-flogics, 2023-03-01:
@paralin ,
Thanks again too!  BTW, please let me add that G++ version 8 is also required.

Atsushi

commit 22effa2a (2023-03-01): virt/qemu: add note about gcc 8 requirement

Fixes #274