skiffos/SkiffOS #275: make cmd/virt/qemu/run failed for intel/x64,virt/qemu

yokoyama-flogics, 2023-03-02:
Hello,

First, thank you for your help with [my question](https://github.com/skiffos/SkiffOS/issues/274) yesterday.

By your help, I could successfully `make compile` for `intel/x64,virt/qemu`.
However, I experienced another problem as following.

FYI, I'm evaluating commit 5efaf68.

When I ran `make cmd/virt/qemu/run`, qemu-system (for x86_64) said it doesn't accept `-machine virt`.
In addition, it also refused `-bios default`.

By removing the above two options, I could successfully launch SkiffOS on qemu.

However, I found yet another problem when trying to login by root.
It didn't ask password (it means password is empty), but after the login, it showed login prompt again.
I mean I can't get a shell prompt.  Please refer below.

```
Welcome to SkiffOS 2022.11.1-98-g5efaf68a!
skiffos-5efaf68a login: [  136.736773] systemd-hostnam (500) used greatest stact

skiffos-5efaf68a login: root
Last login: Thu Mar  2 01:05:24 on ttyS0

Welcome to SkiffOS 2022.11.1-98-g5efaf68a!
skiffos-5efaf68a login:
```

I also tried `SKIFF_CONFIG=intel/x64,virt/qemu,util/rootlogin` but no luck.  (I guess, virt/qemu automatically add `util/rootlogin`.

For your reference purpose, I will also attach kernel log.  Sorry for the long post.

Any suggestion would be appreciated.

Regards,

Atsushi

```
[    0.000000] Linux version 6.1.14 (yokoyama@ubuntu18) (x86_64-buildroot-linux3
[    0.000000] Command line: console=ttyS0 console=tty root=/dev/ram0 crashkernM
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] signal: max sigframe size: 1440
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffdcfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffdd000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000fd00000000-0x000000ffffffffff] reserved
[    0.000000] random: crng init done
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000009fbff] e
[    0.000000] reserve setup_data: [mem 0x000000000009fc00-0x000000000009ffff] d
[    0.000000] reserve setup_data: [mem 0x00000000000f0000-0x00000000000fffff] d
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000000e317df] e
[    0.000000] reserve setup_data: [mem 0x0000000000e317e0-0x0000000000e317ef] e
[    0.000000] reserve setup_data: [mem 0x0000000000e317f0-0x000000007ffdcfff] e
[    0.000000] reserve setup_data: [mem 0x000000007ffdd000-0x000000007fffffff] d
[    0.000000] reserve setup_data: [mem 0x00000000fffc0000-0x00000000ffffffff] d
[    0.000000] reserve setup_data: [mem 0x000000fd00000000-0x000000ffffffffff] d
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS rel-1.16.1-0-g4
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.000000] tsc: Detected 3392.433 MHz processor
[    0.009298] last_pfn = 0x7ffdd max_arch_pfn = 0x400000000
[    0.010269] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT
[... 510 lines cut ...]
[   92.399795] overlayfs: upper fs does not support RENAME_WHITEOUT.
[   92.400556] overlayfs: failed to set xattr on upper
[   92.772617] systemd[1]: Reloading.
[   94.906713] systemd[1]: Reloading.
[   96.842171] systemd[1]: systemd-udev-trigger.service: Deactivated successful.
[   96.852880] systemd[1]: Stopped Coldplug All udev Devices.
[   96.863520] systemd[1]: Stopping Coldplug All udev Devices...
[   96.897973] systemd[1]: Starting Coldplug All udev Devices...
[   96.952259] systemd[1]: Finished SkiffOS early-init.
[   97.010167] systemd[1]: Starting Journal Service...
[   97.015951] systemd[1]: Load Kernel Modules was skipped because no trigger c.
[   97.085743] systemd[1]: Starting Apply Kernel Variables...
[   97.451638] systemd[1]: Finished Apply Kernel Variables.
[   97.757835] systemd[1]: Started Journal Service.
[   97.950817] systemd-journald[417]: Received client request to flush runtime .
[  102.075132] zram: Added device: zram0
[  103.718180] cfg80211: Loading compiled-in X.509 certificates for regulatory e
[  103.908400] powernow_k8: Power state transitions not supported
[  103.912597] powernow_k8: Power state transitions not supported
[  104.262774] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  104.658429] zram0: detected capacity change from 0 to 4194304
[  104.771077] systemd-journald[417]: Data hash table of /var/log/journal/12f50.
[  104.867040] systemd-journald[417]: /var/log/journal/12f50a59191c4351a8a38ae7.
[  105.251261] Adding 2097148k swap on /dev/zram0.  Priority:-2 extents:1 acrosS
```

paralin, 2023-03-02:
Hi, thanks again for the report. It seems to me that bash is crashing after you log in.

@maxberger reported recently that bash was segmentation faulting on riscv, I wonder if that is related.

I'll test it out now and see if I can reproduce.

paralin, 2023-03-02:
Reproduced the issue, working on a fix. Should be ready soon!

paralin, 2023-03-02:
Hey @yokoyama-flogics it's fixed as of master dabaa628a8bbe1f57f652ef7695174cd2d978df5

You will need to recompile - before running "make configure" run "make clean" and select "y" for clearing the compiler cache as well.

Tested it with the configuration you provided and everything seems to be working now.

Unrelated, but I'd definitely recommend enabling Docker / skiff core:

```
# enable docker
export SKIFF_CONFIG=intel/x64,virt/qemu,apps/docker
# enable skiff core (docker + debian linux)
# you can access debian with "su - core"
export SKIFF_CONFIG=intel/x64,virt/qemu,skiff/core
```

Please feel free to let me know if you have any further difficulties & I'll re-open this.

Thanks again for testing!

yokoyama-flogics, 2023-03-02:
@paralin ,

Thank you for the hard work!
I pulled the latest update and now compiling the code.
I also added `skiff/core` to the `SKIFF_CONFIG`.

Atsushi

yokoyama-flogics, 2023-03-02:
@paralin ,

Now I could login to the SkiffOS on qemu!
And also `su - core` ran successfully.

I will evaluate SkiffOS on qemu for a while.
Thank you!!

Atsushi

paralin, 2023-03-02:
@yokoyama-flogics good to hear! One last suggestion; id definitely recommend using ssh to access the VM instead of the serial terminal. The serial terminal has limited size and colors.

Best regards, Christian

commit dabaa628 (2023-03-02): virt/qemu: fix mounting, arch, kernel opts, execute script

Fixes #275
closed 2023-03-02 by paralin