skiffos/SkiffOS #276: SSH root login fails depending on environment variable LANG

yokoyama-flogics, 2023-03-03:
Hello,

I'm afraid of frequent questions.

I found a curious problem.
By @paralin 's suggestion, I'm evaluating ssh login from a remote host.
I found that `ssh core@skiffos_host` works fine, but `ssh root@skiffos_host` fails.
When I said "fail", the login itself goes well, but promptly it exists as following.

```bash
remote_host$ ssh root@skiffos_host
Connection to skiffos_host closed.
remote_host$ 
```

Curiously, the following works fine.

```bash
remote_host$ ssh root@skiffos_host ls -a
.
..
.bash_history
.ssh
```

After some investigation, I found that environment variable `LANG` may cause the login failure.
Exactly speaking, `LANG=C ssh root@skiffos_host` works fine, but `LANG=ja_JP.UTF-8 ssh root@skiffos_host` fails.

FYI, SkiffOS commit is `dabaa62` again.

Does anybody know the reason?

Regards,
Atsushi

paralin, 2023-03-03:
@yokoyama-flogics This must be the same issue @maxberger has on riscv. I think it's because the full list of locales is not included in the target system. Will look into this.

yokoyama-flogics, 2023-03-03:
@paralin ,
Thank you your reply.
I understood that.  First, I didn't understand why bash exits depending `LANG` environment variable. :)
Atsushi

paralin, 2023-03-03:
[image]

If I set just LANG it works, but if i set both LANG and LC_ALL I can reproduce the error.

Checking into how I can either make Bash work when the locale is not found OR include all of the locales.

paralin, 2023-03-03:
@yokoyama-flogics @maxberger I have found this line in sshd_config which enables accepting the LANG environment variables, and commented it out. Now sshd should drop those variables automatically.

For this to apply, you'll have to checkout `master` and run `make compile` (no need to do a full rebuild), for virt/qemu the update will then apply next time you run "make cmd/virt/qemu/run" -

However - the sshd configuration is copied to persist on first boot. So you will also, on any existing system, edit `/mnt/persist/skiff/ssh/sshd_config` and comment out that line as well.

Change: https://github.com/skiffos/SkiffOS/commit/35e66718c668584ca1db4b55b3a73e39d3b17dd5

Let me know if that works!

yokoyama-flogics, 2023-03-03:
@paralin ,
Confirmed that the problem has gone!
Thanks!!

paralin, 2023-03-03:
@yokoyama-flogics glad to hear!

@maxberger hopefully that fixes the riscv issue too.

closed 2023-03-03 by paralin