skiffos/SkiffOS #285: Build fails for lichee RV

hansfbaier, 2023-04-17:
```
[17:44:22] ~/HDL/MiSTeX/SkiffOS master|✓$ echo $SKIFF_CONFIG
allwinner/licheerv,skiff/core
[17:44:33] ~/HDL/MiSTeX/SkiffOS master|✓$ make compile
make[1]: Entering directory '/devel/HDL/MiSTeX/SkiffOS/build'
+ '[' '!' -d /home/jack/HDL/MiSTeX/SkiffOS/workspaces/default ']'
+ '[' '!' -f /home/jack/HDL/MiSTeX/SkiffOS/workspaces/default/Makefile ']'
+ '[' '!' -d /home/jack/HDL/MiSTeX/SkiffOS/overrides/workspaces/default/ ']'
+ ln -f -s /home/jack/HDL/MiSTeX/SkiffOS/workspaces/default /home/jack/HDL/MiSTeX/SkiffOS/workspaces/default/output
Skiff config matches allwinner/d1,skiff/moduleimg,allwinner/licheerv,apps/runc,apps/containerd,apps/docker,skiff/core, skipping config rebuild.
Kicking off Buildroot build...
cd $BUILDROOT_DIR && make 
make[2]: Entering directory '/devel/HDL/MiSTeX/SkiffOS/workspaces/default'
***
*** Can't find default configuration "arch/../configs/lichee_rv_defconfig"!
***
make[5]: *** [scripts/kconfig/Makefile:96: lichee_rv_defconfig] Error 1
make[4]: *** [Makefile:575: lichee_rv_defconfig] Error 2
make[3]: *** [boot/uboot/uboot.mk:553: /devel/HDL/MiSTeX/SkiffOS/workspaces/default/build/uboot-custom/.stamp_dotconfig] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/devel/HDL/MiSTeX/SkiffOS/workspaces/default'
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/devel/HDL/MiSTeX/SkiffOS/build'
make: *** [Makefile:2: compile] Error 2
[17:44:41] ~/HDL/MiSTeX/SkiffOS master|✓$
```

paralin, 2023-04-17:
Thanks for the report, I'll fix today

paralin, 2023-04-17:
@hansfbaier I ran a build on `master` and it worked fine. I suspect that you have run a build for a different board before licheerv and then switched + tried to build without cleaning first.

Please run these steps:

```
git checkout master
git fetch origin
git reset --hard origin/master
git submodule update
rm -rf ./workspaces/default
export SKIFF_WORKSPACE=default
export SKIFF_CONFIG=allwinner/licheerv,skiff/core
make compile
```

If you are still having issues feel free to reply here and I'll help.

paralin, 2023-04-17:
@hansfbaier Ah I see this was an issue building uboot and not linux. Looking into it now.

paralin, 2023-04-17:
@hansfbaier Okay thanks again for the report, fixed in 7163c4d

closed 2023-04-17 by paralin

hansfbaier, 2023-04-17:
> @hansfbaier <https://github.com/hansfbaier> Please run these steps:
>
> git pull
> git submodule update
> make br/uboot-dirclean
> make configure compile

That worked, thanks!

hansfbaier, 2023-04-17:
@paralin That worked, thanks!