skiffos/SkiffOS #286: container fails to initialize because of TLS certificate verification failure

hansfbaier, 2023-04-18:
First time I want to log in as user `core`:
```
[14:57:33] ~ $ ssh core@192.168.1.30
Container setup in progress:
Step 1/15 : ARG DISTRO
Step 2/15 : FROM ${DISTRO:-debian:sid} as stage1
Image setup failed with error:
Get "https://registry-1.docker.io/v2/": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: current time 2022-12-20T18:29:07Z is before 2023-02-22T00:00:00Z

2022/12/20 18:29:29 Re-opening truncated file /home/core/.skiff-core-setup.log ...
2022/12/20 18:29:29 Successfully reopened truncated /home/core/.skiff-core-setup.log
Step 1/15 : ARG DISTRO
Step 2/15 : FROM ${DISTRO:-debian:sid} as stage1
Image setup failed with error:
Get "https://registry-1.docker.io/v2/": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: current time 2022-12-20T18:29:32Z is before 2023-02-22T00:00:00Z
```

hansfbaier, 2023-04-18:
Hmm the default ntp server which is running does not seem to get the date.
Manually with ntpdate it seemed to work,
then it is no problem.

paralin, 2023-04-18:
@hansfbaier it seems to work fine on my machines & I've not had people report this particular issue before. Perhaps it was a temporary issue with ntp? Or maybe something like the default ntp server is blocked in your environment?

If you could reset the device (clear persist) and boot again, check if it happens again, if so, please send me a .tar.gz of persist/skiff/journal

hansfbaier, 2023-04-18:
I think the NTP servers probably aren't reachable from here (Indonesia). Only when I used the indonesian one ntpdate worked .

paralin, 2023-04-22:
hey @hansfbaier feel free to reply here if you still need assistance w/ this.

closed 2023-04-22 by paralin