skiffos/SkiffOS #291: VisionFive 2 uboot broken and kernel panic

clayauld, 2023-05-26:
I have been able to manually trigger a boot of SkiffOS through some uboot commands that I referred back to when working on the Visionfive 1 board. 

I used a workaround where I have a working Debian image on eMMC to trigger a successful launch of OpenSBI and U-Boot. I halt the autoboot process and manually enter these commands:

```bash
StarFive # mmc list                                                                                       │
sdio0@16010000: 0 (eMMC)                                                                                  │
sdio1@16020000: 1

StarFive # load mmc 1:1 $kernel_addr_r /boot/Image
22048768 bytes read in 926 ms (22.7 MiB/s)

StarFive #  load mmc 1:1 $fdt_addr_r /boot/jh7110-visionfive-v2.dtb
47809 bytes read in 9 ms (5.1 MiB/s)

StarFive # booti $kernel_addr_r - $fdt_addr_r
```

The kernel then boots, but ends in a kernel panic.

The full U-Boot log is here:

```
Main section boot fail,use backup section

U-Boot SPL 2021.10 (Feb 24 2023 - 03:37:30 -0500)
DDR version: dc2e84f0.
Trying to boot from MMC1

OpenSBI v1.2
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : StarFive VisionFive V2
Platform Features         : medeleg
Platform HART Count       : 5
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 4000000Hz
Platform Console Device   : uart8250
Platform HSM Device       : jh7110-hsm
Platform PMU Device       : ---
Platform Reboot Device    : pm-reset
Platform Shutdown Device  : pm-reset
Firmware Base             : 0x40000000
Firmware Size             : 292 KB
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 1
Domain0 HARTs             : 0*,1*,2*,3*,4*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff (I)
Domain0 Region01          : 0x0000000040000000-0x000000004007ffff ()
Domain0 Region02          : 0x0000000000000000-0xffffffffffffffff (R,W,X)
Domain0 Next Address      : 0x0000000040200000
Domain0 Next Arg1         : 0x0000000042200000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 1
Boot HART Domain          : root
Boot HART Priv Version    : v1.11
Boot HART Base ISA        : rv64imafdcbx
Boot HART ISA Extensions  : none
Boot HART PMP Count       : 8
[... 280 lines cut ...]
[   58.705159] Unable to handle kernel paging request at virtual address ffffffe07ff10000
[   58.713066] Oops [#1]
[   58.715376] Modules linked in:
[   58.718499] CPU: 0 PID: 1 Comm: swapper/0 Tainted: G             L    5.15.0 #1
[   58.725863] Hardware name: StarFive VisionFive V2 (DT)
[   58.731063] epc : crc32_body+0x2c/0x132
[   58.734963]  ra : crc32_be+0x40/0x6a
[   58.738603] epc : ffffffff8036332c ra : ffffffff803634ae sp : ffffffd00400bd90
[   58.745883]  gp : ffffffff815061f0 tp : ffffffe0bff40000 t0 : ffffffffa0ce67de
[   58.753163]  t1 : 0000000000000004 t2 : ffffffff82457406 s0 : ffffffd00400bda0
[   58.760443]  s1 : 000000000000ff00 a0 : ffffffe07ff0fffc a1 : 0000000000001000
[   58.767723]  a2 : ffffffff80e438fc a3 : ffffffff80e43500 a4 : 0000000044c4da5d
[   58.775003]  a5 : ffffffffc924bf4e a6 : ffffffe0d4acccc4 a7 : 0000000000002000
[   58.782283]  s2 : 0000000000ff0000 s3 : 0000000000000007 s4 : ffffffff81507008
[   58.789563]  s5 : ffffffff80c1c038 s6 : ffffffff81507038 s7 : ffffffff80c00478
[   58.796843]  s8 : 0000000000000008 s9 : ffffffff80a000ac s10: 0000000000000000
[   58.804123]  s11: 0000000000000000 t3 : 000000007d192cb7 t4 : fffffffff76765b0
[   58.811403]  t5 : 000000007ba0608f t6 : ffffffffb440f7b1
[   58.816776] status: 0000000200000120 badaddr: ffffffe07ff10000 cause: 000000000000000d
[   58.824751] [<ffffffff8036332c>] crc32_body+0x2c/0x132
[   58.829982] ---[ end trace 365db12a39ca846f ]---
[   58.834636] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[   58.842344] SMP: stopping secondary CPUs
[   58.846347] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---
```

paralin, 2023-05-26:
Unfortunately I don't have a visionfive2, @maxberger were you able to get it working with your board?

I'll take a pass at updating everything related to the config to match the reference buildroot the vendor publishes here shortly (in the next hour or so)

maxberger, 2023-05-26:
I haven't tried with the latest version, will do that tomorrow. I've always used SD card only, so I have no experience with MMC.

clayauld, 2023-05-26:
There are workarounds to getting boot from eMMC working, and the Starfive buildroot build works with that. 

I'm not as concerned with making eMMC boot work until I get this to work on SD. Lots of the distros don't support it yet for this board. That's something for another day....

paralin, 2023-05-26:
@clayauld I updated the kernel and u-boot on master, could you try these steps?

1. Clean with `make clean` (or delete workspaces/visionfive2)
2. Enable bootlin toolchain with `mkdir ./overrides/workspaces/visionfive2/buildroot && echo "BR2_TOOLCHAIN_EXTERNAL=y" > ./overrides/workspaces/visionfive2/buildroot/toolchain`
3. `git pull` master and `git submodule update`
4. Compile the system & flash & test

They seem to have changed some things. The kernel errors you sent above seem to be related to the L2 cache being configured incorrectly. I think their newer versions might fix this. Using the bootlin toolchain is just for testing to speed up the build / remove potential variance in the toolchain.

paralin, 2023-05-26:
Added +1 commit to rebase the kernel patches, changes:

 - Update kernel: bf45b644d66ef71f27fc046a6279a73effed7a4d
 - Update u-boot: 991897316f20fbca5d186356901361f6bd777325
 - Drop u-boot patch: f30fa4d196ce973af28c263e5cb475fe1fc00fe4
 - Rebase kernel patch: ddb1a072cd89c46bf70e84cd026b953d2ab3e7be

maxberger, 2023-05-27:
The previous version I checked and which worked was 2022.11.1-104-gc444c4f0 (the actual hash is without the g)

I have now done a clean checkout and build of the current version, 2023.02.1-25-gddb1a072 (hash: ddb1a072cd89c46bf70e84cd026b953d2ab3e7be). This one boots and starts fine on my visionfive2.

my buildscript (non-root):
```
export SKIFF_WORKSPACE=starfive
export SKIFF_CONFIG=starfive/visionfive2,apps/docker,apps/compose
make configure
make compile
```

then as root:
```
export SKIFF_WORKSPACE=starfive
export SKIFF_CONFIG=starfive/visionfive2,apps/docker,apps/compose
export STARFIVE_IMAGE=starfive.img
make cmd/starfive/common/buildimage
chmod 666 starfive.img

```

I then write the sd image to an sd card, the visionfive boots directly from there, no mmc. (bootswitches are at 3 and 4).

The relevant dmesg lines after your crash:
```
[    2.019094] lib80211_crypt: registered algorithm 'NULL'
[    2.019110] lib80211_crypt: registered algorithm 'WEP'
[    2.019124] lib80211_crypt: registered algorithm 'CCMP'
[    2.019138] lib80211_crypt: registered algorithm 'TKIP'
[    2.019174] Key type dns_resolver registered
[    2.024761] Loading compiled-in X.509 certificates
[    2.071100] starfive_jh7110-pinctrl 13040000.gpio: SiFive GPIO chip registered 64 GPIOs
[    2.080757] starfive_jh7110-pinctrl 17020000.gpio: SiFive GPIO chip registered 4 GPIOs
[    2.089720] pl08xdmac 16008000.sec_dma: initialized 8 virtual memcpy channels
[    2.097608] pl08xdmac 16008000.sec_dma: initialized 16 virtual slave channels
[    2.107282] debugfs: Directory '16008000.sec_dma' with parent 'dmaengine' already present!
[    2.116407] pl08xdmac 16008000.sec_dma: DMA: PL080 rev0 at 0x16008000 irq 23
[    2.124436] ssp-pl022 10060000.spi: ARM PL022 driver for StarFive SoC platform, device ID: 0x00041022
[    2.134621] ssp-pl022 10060000.spi: mapped registers from 0x0000000010060000 to (____ptrval____)
[    2.144739] ssp-pl022 10060000.spi: Requested frequency: 10000000 Hz is unsupported,select by default 8250000 Hz
[    2.156356] ssp-pl022 10060000.spi: will use autosuspend for runtime pm, delay 100ms
[    2.166289] i2c 2-0045: Fixing up cyclic dependency with 295d0000.mipi
[    2.173780] seeed_panel 2-0045: Unknown Atmel firmware revision: 0x00
[    2.181085] tinker_ft5406 2-0038: Address = 0x38
[    2.186163] tinker_ft5406 2-0038: width = 800, height = 480, reverse = 1
[    2.193687] tinker_ft5406 2-0038: i2c read error, -121
[    2.199361] tinker_ft5406 2-0038: Checking touch ic failed
[    2.205399] tinker_ft5406: probe of 2-0038 failed with error -121
[    2.212318] i2c 2-0019: Fixing up cyclic dependency with 295d0000.mipi
[    2.220436] Goodix-TS 2-0014: GTP Driver Version: V2.2<2014/01/14>
[    2.227280] Goodix-TS 2-0014: supply tp not found, using dummy regulator
[    2.437857] Goodix-TS 2-0014: I2C Read: 0x8000, 10 bytes failed, errcode: -121! Process reset.
[    2.607852] Goodix-TS 2-0014: I2C Read: 0x8000, 10 bytes failed, errcode: -121! Process reset.
[    2.777187] Goodix-TS 2-0014: Failed to get chip-type
[    2.782738] Goodix-TS 2-0014: Get chip type failed.
[    2.788254] Goodix-TS: probe of 2-0014 failed with error -22
[    2.795406] at24 5-0050: supply vcc not found, using dummy regulator
[    2.803276] at24 5-0050: 512 byte 24c04 EEPROM, writable, 16 bytes/write
[    2.812627] axp15060-regulator 5-0036: Register mipi_0p9 done! vol range:900 ~ 900 mV
[    2.822768] axp15060-regulator 5-0036: Register hdmi_1p8 done! vol range:1800 ~ 1800 mV
[    2.833131] axp15060-regulator 5-0036: Register hdmi_0p9 done! vol range:900 ~ 900 mV
[    2.843305] axp15060-regulator 5-0036: Register cpu_vdd done! vol range:500 ~ 1540 mV
```

-> I cannot replicate the error.

maxberger, 2023-05-27:
Possible next steps:
*  Try and see if you can do it completely without EMMC
* Ubuntu has recently been released with kernel 6.2 ( https://wiki.ubuntu.com/RISC-V/StarFive%20VisionFive%202 ); we could try to use their kernel.
* You may have a different board revision (unfortunately I don't know and can't see mine, but I have the 8GB model with both 1000 MBit ports, so i think it is 1.3b). In this case, the device tree may need to be modified. 

I will be away from the machine for 2 weeks, and since I can't replicate the issue I will probably be of little help here. Once I am back I could try to work on the 6.2 kernel if you haven't done so yet.

paralin, 2023-05-27:
@clayauld how much memory does your board have? I believe there's a patch that assumes 8GB

clayauld, 2023-05-27:
My board is the 8GB version. I have built the new kernel successfully, just haven't had a chance to try booting the board yet. I may be able to get to that this afternoon.

clayauld, 2023-05-29:
Again, I had to boot from eMMC with valid Debian image, interrupt the boot process, and manually enter the following:

```
load mmc 1:1 $kernel_addr_r /boot/Image
load mmc 1:1 $fdt_addr_r /boot/jh7110-visionfive-v2.dtb
booti $kernel_addr_r - $fdt_addr_r
```

Full boot log:

```
Main section boot fail,use backup section

U-Boot SPL 2021.10 (Feb 24 2023 - 03:37:30 -0500)
DDR version: dc2e84f0.
Trying to boot from MMC1

OpenSBI v1.2
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : StarFive VisionFive V2
Platform Features         : medeleg
Platform HART Count       : 5
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 4000000Hz
Platform Console Device   : uart8250
Platform HSM Device       : jh7110-hsm
Platform PMU Device       : ---
Platform Reboot Device    : pm-reset
Platform Shutdown Device  : pm-reset
Firmware Base             : 0x40000000
Firmware Size             : 292 KB
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 1
Domain0 HARTs             : 0*,1*,2*,3*,4*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff (I)
Domain0 Region01          : 0x0000000040000000-0x000000004007ffff ()
Domain0 Region02          : 0x0000000000000000-0xffffffffffffffff (R,W,X)
Domain0 Next Address      : 0x0000000040200000
Domain0 Next Arg1         : 0x0000000042200000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 1
Boot HART Domain          : root
Boot HART Priv Version    : v1.11
Boot HART Base ISA        : rv64imafdcbx
Boot HART ISA Extensions  : none
Boot HART PMP Count       : 8
Boot HART PMP Granularity : 4096
Boot HART PMP Address Bits: 34
Boot HART MHPM Count      : 2
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

U-Boot 2021.10 (Feb 24 2023 - 03:37:30 -0500)

CPU:   rv64imacu
Model: StarFive VisionFive V2
DRAM:  8 GiB
MMC:   sdio0@16010000: 0, sdio1@16020000: 1
[... 514 lines cut ...]
[   12.360281] Unable to handle kernel paging request at virtual address ffffffe07ff10000
[   12.368186] Oops [#1]
[   12.370496] Modules linked in:
[   12.373618] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.15.0 #1
[   12.379596] Hardware name: StarFive VisionFive V2 (DT)
[   12.384796] epc : crc32_body+0x2c/0x132
[   12.388696]  ra : crc32_be+0x40/0x6a
[   12.392336] epc : ffffffff8036332c ra : ffffffff803634ae sp : ffffffd00400bd90
[   12.399616]  gp : ffffffff815053c0 tp : ffffffe0bff40000 t0 : 0000000018253178
[   12.406896]  t1 : 0000000000000004 t2 : 000000006db23526 s0 : ffffffd00400bda0
[   12.414176]  s1 : 000000000000ff00 a0 : ffffffe07ff0fffc a1 : 0000000000001000
[   12.421456]  a2 : ffffffff80e4393c a3 : ffffffff80e43540 a4 : fffffffffb9a448a
[   12.428736]  a5 : ffffffffc924bf4e a6 : ffffffe0d4acccc4 a7 : 0000000000002000
[   12.436016]  s2 : 0000000000ff0000 s3 : 0000000000000007 s4 : ffffffff81507008
[   12.443296]  s5 : ffffffff80c1c048 s6 : ffffffff81507038 s7 : ffffffff80c00478
[   12.450576]  s8 : 0000000000000008 s9 : ffffffff80a000ac s10: 0000000000000000
[   12.457856]  s11: 0000000000000000 t3 : 000000007d192cb7 t4 : ffffffffe0f3c4b3
[   12.465136]  t5 : 000000007ba0608f t6 : ffffffffb440f7b1
[   12.470509] status: 0000000200000120 badaddr: ffffffe07ff10000 cause: 000000000000000d
[   12.478483] [<ffffffff8036332c>] crc32_body+0x2c/0x132
[   12.483719] ---[ end trace 6305806158a030ea ]---
[   12.488369] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[   12.496076] SMP: stopping secondary CPUs
[   12.500080] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---
```

paralin, 2023-05-30:
@clayauld Since U-Boot is stored inside the SPI flash on board, and opensbi as well,

Could you try updating to the versions that the vendor provides: https://github.com/starfive-tech/VisionFive2#appendix-iii-updating-spl-and-u-boot-binaries-under-u-boot

@maxberger I noticed that the visionfive2 config does not build opensbi, isn't it needed? Seems so from the visionfive2 repo linked above.

clayauld, 2023-05-30:
I have updated U-Boot and OpenSBI to the latest version already based on the repo you linked here.

paralin, 2023-05-30:
@maxberger maybe the patch to change the dts to 8gb is not needed with the newer opensbi?

@clayauld You could try to build the kernel without that patch:

```
rm configs/starfive/visionfive2/kernel_patches/0002-Enable-8gb-of-memory.patch 
make configure br/linux-dirclean compile
```

The kernel oops is saying that it's a page fault error, so I think it's trying to access a memory address out-of-bounds.

paralin, 2023-05-30:
I think u-boot might be responsible for setting some memory related values.

Looking at the u-boot source I see:

```
CONFIG_BOOTCOMMAND="run load_vf2_env;run importbootenv;run boot2; run load_distro_uenv;run distro_bootcmd"
```

Extra env: https://github.com/starfive-tech/u-boot/blob/ac0ac696256abf412826d74ee918dd417e207d7b/include/configs/starfive-visionfive2.h#L242

Possibly running the boot script manually instead of running with the default `run distro_bootcmd` might miss setting these?

paralin, 2023-05-30:
@clayauld Is my understanding correct that you need to run u-boot manually since the SkiffOS u-boot is not booting correctly?

Could you try flashing the working u-boot to the skiffos sd card?

I'll have a look at synchronizing the u-boot build config to the one in the visionfive2 repo as well.

This is a bit hard for me to fix because I don't have a visionfive2 board on hand at the moment.

clayauld, 2023-05-30:
> @maxberger maybe the patch to change the dts to 8gb is not needed with the newer opensbi?
>
> @clayauld You could try to build the kernel without that patch:
>
> ```
> rm [...]

Running the build after doing the above commands.

> @clayauld Is my understanding correct that you need to run u-boot manually since the SkiffOS u-boot is not booting correctly?
>
> Could you try flashing the working u-boot to the skiffos sd card?
>
> I'll [...]

Yes, I am running the u-boot commands manually to attempt the boot. I'll work on getting u-boot flashed to the sd card. It's possible that the u-boot environments are not being set up properly with my manual commands. I was also thinking of attempting to pull in the u-boot build config from the visionfive2 repo myself and testing that out. I can report tomorrow on my progress.

clayauld, 2023-05-30:
Pulling from the Starfive buildroot repo I tried booting with the following commands:

```
setenv fileaddr 0xa0000000
setenv fdtaddr 0x46000000
setenv kernel_addr_r 0x40200000
setenv irdaddr 0x46100000
setenv irdsize 0x5f00000

load mmc 1:1 $kernel_addr_r /boot/Image
load mmc 1:1 $fdt_addr_r /boot/u-boot.dtb
booti $kernel_addr_r - $fdt_addr_r
```

I was able to keep from getting a kernel panic, but I do have a system halt at the last line. Here are the logs:

```
Main section boot fail,use backup section
Main section boot fail,use backup section

U-Boot SPL 2021.10 (Feb 24 2023 - 03:37:30 -0500)
DDR version: dc2e84f0.
Trying to boot from MMC1

OpenSBI v1.2
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : StarFive VisionFive V2
Platform Features         : medeleg
Platform HART Count       : 5
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 4000000Hz
Platform Console Device   : uart8250
Platform HSM Device       : jh7110-hsm
Platform PMU Device       : ---
Platform Reboot Device    : pm-reset
Platform Shutdown Device  : pm-reset
Firmware Base             : 0x40000000
Firmware Size             : 292 KB
Runtime SBI Version       : 1.0

Domain0 Name              : root
Domain0 Boot HART         : 1
Domain0 HARTs             : 0*,1*,2*,3*,4*
Domain0 Region00          : 0x0000000002000000-0x000000000200ffff (I)
Domain0 Region01          : 0x0000000040000000-0x000000004007ffff ()
Domain0 Region02          : 0x0000000000000000-0xffffffffffffffff (R,W,X)
Domain0 Next Address      : 0x0000000040200000
Domain0 Next Arg1         : 0x0000000042200000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 1
Boot HART Domain          : root
Boot HART Priv Version    : v1.11
Boot HART Base ISA        : rv64imafdcbx
Boot HART ISA Extensions  : none
Boot HART PMP Count       : 8
Boot HART PMP Granularity : 4096
Boot HART PMP Address Bits: 34
Boot HART MHPM Count      : 2
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109
[... 225 lines cut ...]
[    1.873237] Bluetooth: RFCOMM TTY layer initialized
[    1.878063] Bluetooth: RFCOMM socket layer initialized
[    1.883252] Bluetooth: RFCOMM ver 1.11
[    1.887072] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    1.892429] Bluetooth: BNEP socket layer initialized
[    1.897463] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[    1.903436] Bluetooth: HIDP socket layer initialized
[    1.908631] lib80211: common routines for IEEE802.11 drivers
[    1.914210] lib80211_crypt: registered algorithm 'NULL'
[    1.919514] lib80211_crypt: registered algorithm 'WEP'
[    1.924697] lib80211_crypt: registered algorithm 'CCMP'
[    1.929994] lib80211_crypt: registered algorithm 'TKIP'
[    1.935288] Key type dns_resolver registered
[    1.940257] Loading compiled-in X.509 certificates
[    1.948432] of_cfs_init
[    1.950873] of_cfs_init: OK
[    1.953912] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    1.963263] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    1.970130] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    1.978689] cfg80211: failed to load regulatory.db
[    4.046737] ALSA device list:
[    4.049627]   No soundcards found.
[    4.053121] Warning: unable to open an initial console.
[    4.058497] Waiting for root device ...
```

maxberger, 2023-05-30:
Since it says waiting for root device:  I had to adapt exlinux.conf to get the root device correct, before that I also had the kernel stop at  the same Point:
 https://github.com/skiffos/SkiffOS/blob/master/configs/starfive/visionfive2/resources/boot-scripts/extlinux.conf#L4

No idea if this helps, but it could give you additional parameters to try

clayauld, 2023-05-30:
Hmmm... my extlinux.conf file has the same parameters. 

I still can't figure out why uboot doesn't like the sd card. And no matter what I seem to do it doesn't want to work with the image I'm using. I tried to build the VisionFive2 image right from the upstream buildroot source and I'm getting the same issues with the SD card not being recognized with this board. So it seems like the upstream buildroot source is broken for whatever version of the board I have.

clayauld, 2023-05-30:
I think I found the reason for uboot not liking the SD card image:

[image]

U-boot is hardcoded in this board to look for 3 partitions on the drive. It is coded to boot from the 3rd partition /boot directory.

paralin, 2023-05-30:
@clayauld that is a genimage config which is used to create a sd card image. It doesn't necessarily imply that uboot is only looking in /boot.

However that could still be the case, the code would be elsewhere in U-boot if so.

paralin, 2023-05-30:
@clayauld okay I see the comment now saying it's hard coded. Makes sense.

We can use a 3 partition layout as well, are you able to test it once to see if it works?

clayauld, 2023-05-30:
Absolutely. If you have a quick patch to the install script or image gen script I can test it quickly. I was going to work on a patch myself but you can probably push it quicker than I can.

paralin, 2023-05-30:
@clayauld if that is indeed the issue you would see uboot logs looking for files and not finding them and then falling back to network boot. If uboot doesn't start at all it's a different issue.

paralin, 2023-05-30:
@clayauld According to @maxberger the current version boots & starts fine on his visionfive2.

Given that combined with the fact that your u-boot logs do not show u-boot searching for the third partition, I have to think that the issue is not related to the partition layout after all.

If you still want to test this, I pushed the `starfive-partitions` branch.

clayauld, 2023-05-30:
@paralin Thanks, I'll test it out. 

After finding more documentation online about different builds for this I'm suspecting something else is going on here too. I can't get to a U-Boot prompt unless I have a TF card inserted with a working image and interrupt the boot process. Something definitely seems to be broken with my board since all documents state I should be able to load U-Boot from SPI flash even with no card or eMMC. Makes me think that the SPI flash is not working properly or the ROM on board is doing some sort of check before it starts the boot process.

clayauld, 2023-05-30:
Turns out this is an issue with my board. The board could not load anything from SPI flash and after following a different set of instructions on flashing uboot and SPL binary to the board I was able to get the system to work as expected.

I followed the flashing instructions here: https://wiki.gentoo.org/wiki/Embedded_Handbook/Boards/StarFive_VisionFive_2

[image]

Talk about frustrating....

I now have a successful boot using the SkiffOS master branch and no issues. This whole problem came down to the fact that uboot and the SBL binary were not properly flashed to the board's SPI onboard flash, and the UART bootloader recovery process did not fix the issue no matter how many times I completed it.

closed 2023-05-30 by clayauld

paralin, 2023-05-30:
@clayauld thanks for figuring it out. If you want to send a PR with more documentation for the readme ill merge it right away!

clayauld, 2023-05-30:
Perfect! I'll work up some documentation to help and request a PR with the changes.