skiffos/SkiffOS #292: odroid-hc1 /hc2 specific dts

coelner, 2023-07-07:
Hey,

there is a specific dtb for those boards:
`https://github.com/torvalds/linux/blob/master/arch/arm/boot/dts/samsung/exynos5422-odroidhc1.dts`

and it is missing here:
`https://github.com/skiffos/SkiffOS/blob/master/configs/odroid/xu/buildroot/dts`

it is rather important because the HC1 and HC2 do not have a fan instead the boards uses just a greater heatsink.

paralin, 2023-07-07:
@coelner as far as I know the hc2 and hc1 are compatible with the xu4 dts. I use those boards with the configuration as-is.

Is there 100% a need to have a separate config for them? Isn't it okay to have the fan in the dts despite it not being connected?

Thanks for the issue report!

coelner, 2023-07-07:
they are compatible, but the thermal-zones are different. The trigger points are higher but the hysteresis is longer. And the speed limits are different (1600/1100) to (1800/1200). That would not lead to catastrophic errors, but maybe to a more often low speed state with 600Mhz. 

some smaller things differs too, like hdmi, power button and sound i2c. That is not really import, but maybe we are generating more i/o request at kernel level due searching those?

Me simplistic view was that this dts could be added and manually activated by the user if power regressions occur.

paralin, 2023-07-11:
@coelner There's no way as far as I can tell to detect which board is running in u-boot.

If you look here: https://github.com/skiffos/SkiffOS/commit/c747f760d2ca12cf1a6371d49e50d45148ef8319#diff-2ecea2f0200e12ec4890307f7bc80c02d32317aaf1dd8597be5e730efee152baR84

I added the odroid hc1 dts to the list to be built and added to the boot/ partition and updated u-boot to 2023.07.

I've also added a conditional line which checks board_name and loads the hc1 if the board_name is set to `hc1.`

That said I checked u-boot and `board_name` is not set in recent versions, so it will just default to the xu4 dts.

If you want to load the hc1 dts, uncomment the line in the boot.txt I linked above which sets `board_name` to hc1.

Hopefully this solves the issue! I can also add another config `odroid/hc1` which inherits `odroid/common` and then forces the boot script to use `board_name = hc1` but that is probably unnecessary for most use cases, so let me know if this solves your case as-is.

coelner, 2023-07-12:
cool, I will do a build with this. 
AFAIK armbian does this also this way and define the used dts afterwards. 
using the xu4 dts is not dangerous, but not ideal. 
the only this to check the used board is that there is missing the i2c fan register. But I don't know if you can check this from u-boot. 

Thank you for diving into this!

paralin, 2023-07-19:
@coelner Let me know if you need anything else!

closed 2023-07-19 by paralin