skiffos/SkiffOS #296: Bad Linux ARM64 Image magic! on odroid m1

nadsat, 2023-09-07:
Hi 

I'm getting an error when booting an odroid m1 from a SD disk. I followed the instructions from the [documentation](https://github.com/skiffos/SkiffOS/tree/master/configs/odroid/m1#odroid-m1) and it looks like it worked. Part of boot info:

```
Found U-Boot script /boot.scr                                                                                                                                                                 
1451 bytes read in 4 ms (353.5 KiB/s)                                                                                                                                                         
## Executing script at 00c00000                                                                                                                                                               
SkiffOS for Odroid M1 board odroidm1 starting...                                                                                                                                              
Loading Image...                                                                                                                                                                              
** Unrecognized filesystem type **                                                                                                                                                            
Loading rootfs.cpio.uboot...                                                                                                                                                                  
** Unrecognized filesystem type **                                                                                                                                                            
Loading rk3568-odroid-m1.dtb                                                                                                                                                                  
** Unrecognized filesystem type **                                                                                                                                                            
Fdt Ramdisk skip relocation                                                                                                                                                                   
No misc partition                                                                                                                                                                             
Bad Linux ARM64 Image magic!                                                                                                                                                                  
SCRIPT FAILED: continuing...
```
uboot version:
```
=> version                                                                                                                                                                                    
U-Boot 2017.09 (Mar 28 2023 - 06:53:03 +0000)                                                                                                                                                 
                                                                                                                                                                                              
aarch64-linux-gnu-gcc (Linaro GCC 6.3-2017.02) 6.3.1 20170109                                                                                                                                 
GNU ld (Linaro_Binutils-2017.02) 2.27.0.20161019   
```
 Any advice would be appreciated

paralin, 2023-09-07:
@nadsat hi, thanks for the report. I'll test on my m1 locally within the next hour or so and will let you know when I find the fix.

paralin, 2023-09-15:
Hi @nadsat - I have confirmed that the odroid m1 boots fine on master:

[image]

Please `make clean` and delete the cache, then build again on master.

```
git checkout master
git pull
git submodule update
make clean
export SKIFF_CONFIG=odroid/m1,skiff/core
make configure compile
```

Feel free to open a new issue or reply here if still having problems. Also would appreciate it if you could let me know if it's working for you now.

Thanks!

closed 2023-09-15 by paralin

nadsat, 2023-09-17:
Hi @paralin thanks for your help. It worked.

[image]

paralin, 2023-09-17:
@nadsat great to hear!