skiffos/SkiffOS #303: recursive dependency issue (gnupg/gnupg2)

theCalcaholic, 2024-01-12:
**I just started to try SkiffOS so please bear with me if I missed anything obvious.**

Whenever I try to build any config with SkiffOS, I get a recursive dependency error, e.g.:

```sh
SKIFF_CONFIG=virt/docker make clean configure compile
[...]
Kicking off Buildroot build...
cd $BUILDROOT_DIR && make 
/usr/bin/make -j1  O=/home/user/projects/skiffos/workspaces/default HOSTCC="/usr/lib64/ccache/gcc" HOSTCXX="/usr/lib64/ccache/g++" syncconfig
  GEN     /home/user/projects/skiffos/workspaces/default/Makefile
package/gnupg2/Config.in:17:error: recursive dependency detected!
package/gnupg2/Config.in:17:	symbol BR2_PACKAGE_GNUPG2 is selected by BR2_PACKAGE_GNUPG
package/gnupg/Config.in:1:	symbol BR2_PACKAGE_GNUPG is selected by BR2_PACKAGE_GNUPG2
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"

Your Perl installation is not complete enough; at least the following
modules are missing:

	 ExtUtils::MakeMaker
	 FindBin

make[3]: *** [support/dependencies/dependencies.mk:27: dependencies] Error 1
make[2]: *** [Makefile:23: _all] Error 2
make[1]: *** [Makefile:64: compile] Error 2
make[1]: Leaving directory '/home/user/projects/skiffos/build'
make: *** [Makefile:2: compile] Error 2
```

I tried a few configs, e.g. `pi/3`, `pi/3x64`, `virt/qemu`, always with the same result.

paralin, 2024-01-12:
Welcome!

The recursive dependency issue is not a problem. It's just a warning and it doesn't stop the build.

The actual issue is you're missing perl modules.

sudo apt-get install libextutils-makemaker-perl`.
 
For `FindBin`, it's usually included in the Perl base package, but if it's missing, it might be due to an incomplete Perl installation. You can reinstall Perl using `sudo apt-get install perl`.

Not sure what OS you're using but your perl installation is broken currently.

theCalcaholic, 2024-01-12:
Oh, so I was on the wrong track all along. Thank you!
Turns out my perl installation wasn't broken, it was missing altogether :)

closed 2024-01-12 by theCalcaholic

paralin, 2024-01-12:
@theCalcaholic no problem! Glad you got it figured out, feel free to reach out if you run into anything else or have any questions or feedback.