skiffos/SkiffOS #306: I have 5 different builds that all fail with the same error...

jim255, 2024-02-16:
Not sure where to begin.  I am attempting to build for Beaglew Black, Beagle AI, Banana M2 original, Banana M3, and Wandboard.   the following is the tail end of each compile output to <STDERR>..
All end with same exact error!

==> b_ai.err <==
/usr/include/features-time64.h:26:5: error: #error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
   26 | #   error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
      |     ^~~~~
make[4]: *** [Makefile:263: mksigname] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: *** [Makefile:260: mksiglist] Error 1
make[3]: *** [package/pkg-generic.mk:274: /home/janderso/Projects/SkiffOS/workspaces/beagle_ai/build/sudo-1.9.15p5/.stamp_configured] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[1]: *** [Makefile:64: compile] Error 2
make: *** [Makefile:2: compile] Error 2

==> banana2.err <==
.config:5817:warning: symbol value 'm' invalid for NF_TABLES_IPV6
.config:5960:warning: symbol value 'm' invalid for USB_FOTG210_HCD
.config:5961:warning: symbol value 'm' invalid for USB_FOTG210_UDC
.config:6084:warning: override: reassigning to symbol USB_SERIAL
make[5]: *** No rule to make target 'arch/arm/boot/dts/sun6i-a31s-sinovoip-bpi-m2.dtb'.  Stop.
make[4]: *** [Makefile:1381: sun6i-a31s-sinovoip-bpi-m2.dtb] Error 2
make[3]: *** [package/pkg-generic.mk:283: /home/janderso/Projects/SkiffOS/workspaces/bananam2/build/linux-6.7.4/.stamp_built] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[1]: *** [Makefile:64: compile] Error 2
make: *** [Makefile:2: compile] Error 2

==> banana3.err <==
/usr/include/features-time64.h:26:5: error: #error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
   26 | #   error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
      |     ^~~~~
make[4]: *** [Makefile:263: mksigname] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: *** [Makefile:260: mksiglist] Error 1
make[3]: *** [package/pkg-generic.mk:274: /home/janderso/Projects/SkiffOS/workspaces/bananam3/build/sudo-1.9.15p5/.stamp_configured] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[1]: *** [Makefile:64: compile] Error 2
make: *** [Makefile:2: compile] Error 2

==> bbeagle.err <==
/usr/include/features-time64.h:26:5: error: #error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
   26 | #   error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
      |     ^~~~~
make[4]: *** [Makefile:263: mksigname] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: *** [Makefile:260: mksiglist] Error 1
make[3]: *** [package/pkg-generic.mk:274: /home/janderso/Projects/SkiffOS/workspaces/beagle_black/build/sudo-1.9.15p5/.stamp_configured] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[1]: *** [Makefile:64: compile] Error 2
make: *** [Makefile:2: compile] Error 2

==> wb.err <==
/usr/include/features-time64.h:26:5: error: #error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
   26 | #   error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
      |     ^~~~~
make[4]: *** [Makefile:260: mksiglist] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: *** [Makefile:263: mksigname] Error 1
make[3]: *** [package/pkg-generic.mk:274: /home/janderso/Projects/SkiffOS/workspaces/wandboard/build/sudo-1.9.15p5/.stamp_configured] Error 2
make[2]: *** [Makefile:23: _all] Error 2
make[1]: *** [Makefile:64: compile] Error 2
make: *** [Makefile:2: compile] Error 2

paralin, 2024-02-16:
Hi, thanks for the report. I see the same thing on a few reports on Discord too. CI is green and my builds are passing, so I think it's a incompatibility with certain Linux distros. Should have a fix today.

paralin, 2024-02-17:
I have reproduced the issue on Ubuntu 22.04 LTS:

issue:

```
/usr/bin/make -j9 GIT_DIR=. PATH="/root/skiffos/workspaces/bpi/host/bin:/root/skiffos/workspaces/bpi/host/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin" PKG_CONFIG="/root/skiffos/workspaces/bpi/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/root/skiffos/workspaces/bpi/host/lib/pkgconfig:/root/skiffos/workspaces/bpi/host/share/pkgconfig" AR="/usr/bin/ar" AS="/usr/bin/as" LD="/usr/bin/ld" NM="/usr/bin/nm" CC="/root/skiffos/workspaces/bpi/host/bin/ccache /usr/bin/gcc" GCC="/root/skiffos/workspaces/bpi/host/bin/ccache /usr/bin/gcc" CXX="/root/skiffos/workspaces/bpi/host/bin/ccache /usr/bin/g++" CPP="/usr/bin/cpp" OBJCOPY="/usr/bin/objcopy" RANLIB="/usr/bin/ranlib" CPPFLAGS="-I/root/skiffos/workspaces/bpi/host/include" CFLAGS="-O2 -I/root/skiffos/workspaces/bpi/host/include" CXXFLAGS="-O2 -I/root/skiffos/workspaces/bpi/host/include" LDFLAGS="-L/root/skiffos/workspaces/bpi/host/lib -Wl,-rpath,/root/skiffos/workspaces/bpi/host/lib" INTLTOOL_PERL=/usr/bin/perl CPPFLAGS="-I/root/skiffos/workspaces/bpi/host/include -I../../include -I../.." -C /root/skiffos/workspaces/bpi/build/sudo-1.9.15p5/lib/util mksigname mksiglist
/usr/bin/cpp -I/root/skiffos/workspaces/bpi/host/include -I../../include -I../.. ./sys_signame.h | /usr/bin/sed -e '1,/^int sudo_end_of_headers;/d' -e '/^#/d' > mksigname.h
/usr/bin/cpp -I/root/skiffos/workspaces/bpi/host/include -I../../include -I../.. ./sys_siglist.h | /usr/bin/sed -e '1,/^int sudo_end_of_headers;/d' -e '/^#/d' > mksiglist.h
In file included from /usr/include/features.h:392,
                 from /usr/include/x86_64-linux-gnu/sys/types.h:25,
                 from ./sys_signame.h:4:
/usr/include/features-time64.h:26:5: error: #error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
   26 | #   error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
      |     ^~~~~
In file included from /usr/include/features.h:392,
                 from /usr/include/x86_64-linux-gnu/sys/types.h:25,
                 from ./sys_siglist.h:4:
/usr/include/features-time64.h:26:5: error: #error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
   26 | #   error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
      |     ^~~~~
/root/skiffos/workspaces/bpi/host/bin/ccache /usr/bin/gcc -I../../include -I../.. -I. -I. -D_PATH_SUDO_CONF=\"/etc/sudo.conf\" -I/root/skiffos/workspaces/bpi/host/include -DDEFAULT_TEXT_DOMAIN=\"sudo\" -O2 -I/root/skiffos/workspaces/bpi/host/include ./mksigname.c -o mksigname
/root/skiffos/workspaces/bpi/host/bin/ccache /usr/bin/gcc -I../../include -I../.. -I. -I. -D_PATH_SUDO_CONF=\"/etc/sudo.conf\" -I/root/skiffos/workspaces/bpi/host/include -DDEFAULT_TEXT_DOMAIN=\"sudo\" -O2 -I/root/skiffos/workspaces/bpi/host/include ./mksiglist.c -o mksiglist
In file included from /usr/include/features.h:392,
                 from /usr/include/x86_64-linux-gnu/bits/libc-header-start.h:33,
                 from /usr/include/stdlib.h:26,
                 from ./mksigname.c:27:
/usr/include/features-time64.h:26:5: error: #error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
   26 | #   error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
      |     ^~~~~
In file included from /usr/include/features.h:392,
                 from /usr/include/x86_64-linux-gnu/bits/libc-header-start.h:33,
                 from /usr/include/stdlib.h:26,
                 from ./mksiglist.c:27:
/usr/include/features-time64.h:26:5: error: #error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
   26 | #   error "_TIME_BITS=64 is allowed only with _FILE_OFFSET_BITS=64"
      |     ^~~~~
make[4]: *** [Makefile:263: mksigname] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: *** [Makefile:260: mksiglist] Error 1
make[3]: *** [package/pkg-generic.mk:274: /root/skiffos/workspaces/bpi/build/sudo-1.9.15p5/.stamp_configured] Error 2
```

glibc version:

```
root@ubuntu-c-8-intel-sfo3-01:~/skiffos# /lib/x86_64-linux-gnu/libc.so.6
GNU C Library (Ubuntu GLIBC 2.35-0ubuntu3.6) stable release version 2.35.
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 11.4.0.
```

I think it's an incompatibility with older host glibc (2.35 vs 2.38). Will check further for a fix.

paralin, 2024-02-17:
Found the issue & fixed it on master, please try: `git pull && git submodule update` then try again.

Submitted a e-mail to the mailing list: https://lists.buildroot.org/pipermail/buildroot/2024-February/685845.html

The issue was we need to pass these flags to host gcc: `HOST_CPPFLAGS  = -I$(HOST_DIR)/include -D_FILE_OFFSET_BITS=64 -D_TIME_BITS=64`

Added Ubuntu 22.04 to the CI so we can catch this type of issue in future.

closed 2024-02-17 by paralin

paralin, 2024-02-17:
This seems to be fixed on master now. I also fixed the path to the dts for the bananapi m2.

Please try now, thanks for the issue report.

```
git pull && git submodule update
```

then make again.