skiffos/SkiffOS #323: Writing an image to a file?

symgryph, 2024-11-24:
I managed to get everything working at the compile step, but its on a remote machine without a sd card reader. What is the command to compile to an .img format I can then write on a  different computer?

paralin, 2024-11-24:
Hi @symgryph it depends on which target you're using. For raspberry pi it's `make cmd/pi/common/buildimage` - you can run "make help" and check at the end of the output for the available commands. There's usually also a readme under configs/pi/common (or whatever target you're using) that has that reference too. Let me know if you found it!

symgryph, 2024-11-24:
I did finally manage to make it compile the image. I just attached a virtual device to the virtual machine. One question though, the system wouldn't even recognize it when I tried to boost though, I tried using a default config for an m1. Keep in mind that I also have an nvme drive with an operating system on it and when I told it to boot from the MMC BLK0 it simply didn't do anything.

Sincerely,

THOMAS MUNN

paralin, 2024-11-24:
Odroid m1? I assume you mean you attached a device to your virtual (build) machine and then used format and then install? I have a m1 here, will test.

symgryph, 2024-11-25:
Yes, I have an odroid m1 with mvme drive on it. I did try pushing the recovery but the SD card never loaded. I was using pettiboot with a menu selection. I also tried the recovery button but that didn't seem to work either. There's an operating system on the nvme drive that does boot

Sincerely,

THOMAS MUNN

paralin, 2024-11-27:
@symgryph Working on testing this when I have a moment. My current system, I built and did "format" then "install" directly to a sd card, then booted that. The system is somewhat old though, so I am working on building and testing the latest version.

symgryph, 2024-11-27:
Thank you very much! I would love to run skiffos. Unfortunately, I don't have a serial console so I'm not sure if the output is going to the serial console. It's certainly not going to the HDMI port.

Sincerely,

THOMAS MUNN

paralin, 2024-12-02:
Built the image, will try to test w/ serial debug monitor today.

symgryph, 2024-12-02:
I eagerly await your results

Sincerely,

THOMAS MUNN

paralin, 2024-12-03:
I see you mentioned petitboot, we use u-boot with SkiffOS. So you would need to boot directly to the SD card, try without the other drive connected. I can have a look at what would be needed to make it work with petitboot as well, if you confirm that's the issue. But it probably needs some additional config files for petitboot to start it properly.