skiffos/SkiffOS #325: bananapi/m2ultra: Compile of commit [2a61137] fails at UBOOT with BINMAN error "unrecognized arguments: -o ConnectTimeout=10"

rschueler, 2025-02-07:
uboot 2023.07 Building
...
  COPY    spl/u-boot-spl.bin
  MKIMAGE spl/sunxi-spl.bin
  BINMAN  .binman_stamp
usage: binman [-h] [-B BUILD_DIR] [-D] [-H] [--tooldir TOOLDIR] [--toolpath TOOLPATH] [-T THREADS] [--test-section-timeout] [-v VERBOSITY] [-V]
              {build,bintool-docs,entry-docs,ls,extract,replace,sign,test,tool} ...
binman: **error: unrecognized arguments: -o ConnectTimeout=10**
make[5]:  [Makefile:1111: .binman_stamp] Error 2
make[4]:  [package/pkg-generic.mk:289: /opt/projects/skiff-ext/skiffos/workspaces/bpi/build/uboot-2023.07/.stamp_built] Error 2

paralin, 2025-02-12:
See commit 9ad23e55187663f626d374a94bbd7fdb7c6cdb49

paralin, 2025-02-12:
https://lore.kernel.org/buildroot/96b69abb843dc7292777c64f6787f134@free.fr/

Issue found!

commit 9cc5f8de (2025-02-12): bananapi/common: update u-boot to 2025.01

Fixes #325

Update U-Boot version for BananaPi boards to 2025.01.

* **configs/bananapi/common/buildroot/uboot**
  - Update U-Boot version to 2025.01.

* **README.md**
  - Update U-Boot version for all BananaPi boards to 2025.01.
  

---

For more details, open the [Copilot Workspace session](https://copilot-workspace.githubnext.com/skiffos/SkiffOS/issues/325?shareId=XXXX-XXXX-XXXX-XXXX).
commit 2a60efac (2025-02-12): bananapi/common: fix build failure with uboot

Fixes #325
closed 2025-02-12 by paralin