skiffos/SkiffOS #327: pi/3 Build failed ... rpi-firmware ... hardfp/opt/vc/bin/vcdbg': No such file or directory

rschueler, 2025-02-19:
>> rpi-firmware 6b029c7772b5d29a50a07c477a808a7effb0a373 Installing to target
/usr/bin/install -D -m 0700 /opt/projects/SkiffOS/workspaces/pi3/build/rpi-firmware-6b029c7772b5d29a50a07c477a808a7effb0a373/hardfp/opt/vc/bin/vcdbg /opt/projects/SkiffOS/workspaces/pi3/target/usr/sbin/vcdbg
/usr/bin/install: cannot stat '/opt/projects/SkiffOS/workspaces/pi3/build/rpi-firmware-6b029c7772b5d29a50a07c477a808a7effb0a373/hardfp/opt/vc/bin/vcdbg': No such file or directory
make[3]: *** [package/pkg-generic.mk:380: /opt/projects/SkiffOS/workspaces/pi3/build/rpi-firmware-6b029c7772b5d29a50a07c477a808a7effb0a373/.stamp_target_installed] Error 1
make[2]: *** [Makefile:23: _all] Error 2
make[2]: Leaving directory '/opt/projects/SkiffOS/workspaces/pi3'

paralin, 2025-02-19:
Ah, we are building pi3 in arm not arm64 but upstream raspberry pi dropped all of the 32 bit binaries from the firmware repo. We should probably just make 64 bit arm the default for pi 3. Does the pi 3 64 bit build work?

paralin, 2025-02-20:
Fixed: https://github.com/skiffos/SkiffOS/commit/37a2088a79553eebd7cb092352ac007834cf760b

Thanks for the report.

closed 2025-02-20 by paralin