skiffos/SkiffOS #329: Banana PI M3 build issues

jim255, 2025-03-18:
I'm getting the following error after build while attempting to create bootable media..
can't find u-boot image at /home/janderso/Projects/SkiffOS/workspaces/bananam3u/images/u-boot-sunxi-with-spl.bin
The contents of folder in the statment appear to otherwise be correct, file not found.

ls /home/janderso/Projects/SkiffOS/workspaces/bananam3u/images
ar7010_1_1.fw  ar9170-2.fw  ath9k_htc  carl9170-1.fw  htc_9271.fw  rootfs.cpio.lz4    rt2561s.bin  rt2870.bin  rt3090.bin  rtlwifi
ar7010.fw      ar9271.fw    brcm       cypress        resources    rootfs.cpio.uboot  rt2661.bin   rt3070.bin  rt73.bin    rtw88
ar9170-1.fw    ath6k        bzImage    htc_7010.fw    rootfs.cpio  rt2561.bin         rt2860.bin   rt3071.bin  rtl_bt      skiff-release

 Jim Anderson

paralin, 2025-03-18:
Hey @jim255 thanks for the report. I'll have a look now and fix the build.

paralin, 2025-03-18:
Hi @jim255 it looks like uboot did not build successfully on your build.

I did try it on my end and it seems to have worked.

Can you try these and let me know? And if the "u-boot-sunxi-with-spl" file is not there still, provide the last 10 lines of logs?

```
git pull
git submodule update
make configure
cd ./workspaces/bananapim3u
make uboot-dirclean
make uboot
```

Thanks!

jim255, 2025-03-18:
good to here!
janderso@gze-006:~/Projects/SkiffOS$ cd workspaces/bananam3u/
janderso@gze-006:~/Projects/SkiffOS/workspaces/bananam3u$ make uboot-dirclean
make[1]: *** No rule to make target 'uboot-dirclean'.  Stop.
make: *** [Makefile:23: _all] Error 2
janderso@gze-006:~/Projects/SkiffOS/workspaces/bananam3u$ make uboot
make[1]: *** No rule to make target 'uboot'.  Stop.
make: *** [Makefile:23: _all] Error 2

I'll try again with a complete build tomorrow!

paralin, 2025-03-18:
That seems to suggest that uboot was not configured. Could you provide `cat ./workspaces/.config_bananam3u/skiff_config` output?

jim255, 2025-03-18:
sure.
apps/runc,apps/containerd,apps/docker,skiff/core,core/ubuntu

paralin, 2025-03-18:
Looks like you're missing the bananapi m3 config.

Try:

```
export SKIFF_WORKSPACE=bananam3u
export SKIFF_CONFIG=bananapi/m3,apps/runc,apps/containerd,apps/docker,skiff/core,core/ubuntu
make clean
make compile
```

That should work!

jim255, 2025-03-19:
Ok.. that resulted in full build with no critical errors. However, the image built, WILL NOT BOOT!!!

paralin, 2025-03-19:
Unfortunately my banana pi M3 is sitting in a storage unit in a different state right now. Otherwise I would test the boot and try to debug it for you. Do you have a serial debug cable? That would make it much easier to diagnose the issue

jim255, 2025-03-20:
I'm gonna have to order a new one!

paralin, 2025-04-17:
@jim255 Hi, did you have a chance to get a cable? Would love to get this fixed.